mmakaay/esphome-xiaomi_bslamp2 #2: Buttons on lamp don't work

bonuzzz, 2021-04-06:
Hi
I flashed your firmware on bedside lamp, but buttons and slider don't work. Is it okay for now?

mmakaay, 2021-04-06:
The firmware is still under heavy development. The protocol for the front panel buttons has been reverse engineered, but the code still has to be written. 

Currently I am working on the last bits of the light implementation, so I hope to continue soon with the button implementation.

I will add a warming to the readme about the development state. 
In to he meanwhile, I post updates to the Home Assistant community website:

https://community.home-assistant.io/t/hacking-yeelight-fw-enabling-lan-control/284406/2

closed 2021-04-06 by bonuzzz

mmakaay, 2021-04-06:
I added some extra information to the README.md file about the development status of this project.
Thanks for triggering me to do so, @bonuzzz

mmakaay, 2021-04-14:
A quick update: I have implemented all required components for controlling the lamp.
If you upgrade to the latest code, you can make the buttons work therefore.
There are already some options in the doc/example.yaml that might serve as inspiration.

bonuzzz, 2021-04-14:
Yes, I follow to your work and and update the lamp daily. I've already used button and slider yesterday :) Great work! Thanks