mmakaay/esphome-xiaomi_bslamp2 #6: Binary output that exposes night light mode on/off

mmakaay, 2021-04-14:
It would be useful to know whether or not the night light mode is activated. While playing with the yaml configs, I found that I had to use this construction to detect night light mode (to turn off the front panel illumination when light mode was active):

```yaml
light:
  - platform: yeelight_bs2
    on_brightness:
      then:
        - output.set_level:
            level: !lambda if (x < 0.012f) return 0; else return x;
```

Having to use the magic number `0.012f` in there is not okay.

mmakaay, 2021-04-15:
Well, why binary? I don't know really. The lamp has more modes available.
I added a text sensor, which propagates the active light mode as a string ("off", "rgb", "white" or "night").
Commit: https://github.com/mmakaay/esphome-yeelight_bs2/commit/d1d9ecea95eab8688500f9dd63f494d5558273a7

It's not enough to close this issue. I want an `on_state` automation for this text sensor, so we have a really easy way to act upon the various light modes ("when the lamp goes into night light mode, play the music box MP3 for 15 minutes", that kind off goodness ;-) )

mmakaay, 2021-04-15:
I made it possible to use the text_sensor within logic expressions, so that fixes the magic number issue.
I can now use the following construct:

```yaml
light:
    on_brightness:
      then:
        - output.set_level:
            id: ${id_front_panel_output}
            level: !lambda |-
              if (id(${id_light_mode}).state == "night")
                return 0;
              else
                return x;
```

mmakaay, 2021-04-15:
There's of course an `on_value` automation trigger for text sensors already. I guess I was a bit tired when I wrote that I needed.to build a trigger. 
Case closed!

closed 2021-04-15 by mmakaay

bonuzzz, 2021-04-15:
Does touch buttons can detect twice press or long press? if yes, maybe it will be useful to use for example long press for night light.
And in stock firmware you can choose time period when night light turns on by default on press button, but i'm not sure it could be done in esphome

mmakaay, 2021-04-15:
Yes, this is possible. ESPHome has the `on_multi_click` trigger, which allows you to configure any kind of button touch schema.
From there you can perform any action that is supported by ESPHome.

In fact, the current example.yaml already has an example of turning on the night light mode, by long touching the power button.
Here's [a link to the relevant section of the yaml configuration](https://github.com/mmakaay/esphome-yeelight_bs2/blob/e97a530c033ebed21c3411594a0db8533574f0a2/doc/example.yaml#L131).