mmakaay/esphome-xiaomi_bslamp2 #7: Naming: stick with yeelight_bs2 or should it be xiaomi_bslamp2?

mmakaay, 2021-04-14:
The branding for this lamp is a bit of a mess on the internet, since it's offered both as "Xiaomi Mijia Bedside Lamp 2" and "Yeelight Bedside Lamp 2". When I bought mind, the name "Xiaomi" was nowhere to be found in the ordering process.
Yeelight appears to be the production company, whereas Xiaomi Mijia might be the actual intended brand for the device.
The device firmware identifies as "yeelink.light.bslamp2" BTW, just to make things 

One pro of keeping "Xiaomi", would be that it fits nicely next to other xiaomi components that are already in ESPHome. These are all named like "xiaomi_<name>". So when renaming the component, I would probably go for "xiaomi_bslamp2".

Any thoughts on this?

bonuzzz, 2021-04-15:
I think it's better to rename to xiaomi because this lamp doesn't have anything related to yeelight. It's xiaomi mijia bedside lamp 2

mmakaay, 2021-04-15:
Okay, some info that I gathered:

The lamp's official name is "Mi Bedside Lamp II".
The lamp is made by Yeelight under the MIJIA brand.
The MIJIA brand is Xiaomi's smart home brand. The term "MIJIA" literally means Xiaomi Smart Home in Chinese.
It was established in 2016 and encompasses all the major smart home products released by Xiaomi under the brand.
Looking around, it looks like "Xiaomi MIJIA" is the most regular combination that is used as the brand name.

Based on all this, I would say it's indeed time to do some renaming in the code.
I'll go with "Xiaomi" and not "MIJIA" to be in line with all the other already implemented xiaomi_* components.

mmakaay, 2021-04-15:
Okay, all done. The code has been updated to reflect the correct naming and the repository URL has been updated.
The old URLs will still work. Thanks GitHub, great feature!

closed 2021-04-15 by mmakaay