mmakaay/esphome-xiaomi_bslamp2 #15: [FEATURE] Validate preset names and group names at compile time

mmakaay, 2021-04-18:
**Is your feature request related to a problem? Please describe.**
I can compile a firmware in which I've used `preset.activate` with non-existing template group names or template names.
The firmware will complain at runtime about the errors via the logging. That is not optimal, since users will not continuously be watching the logs. When a user runs into a non-operational `preset.activate` action, the verdict is likely that the code is broken.

**Describe the solution you'd like**
At compile time, check if the arguments for `preset.activate` are valid. If not, provide a clear error message describing the issue.

mmakaay, 2021-04-19:
Improved the preset configuration validation in https://github.com/mmakaay/esphome-xiaomi_bslamp2/commit/e58c4fba3d4e3db97fbc88622fe7b433d4fc00e6
This is what I can do for now. Closing this issue.

mmakaay, 2021-04-19:
Reopening! I was pointed at this PR, which implements what I'd need for handling the checks:
https://github.com/esphome/esphome/pull/1631

closed 2021-08-10 by mmakaay

kroimon, 2022-03-21:
There is still a reference to this issue in `doc/configuration.md` which should probably be removed.

mmakaay, 2022-03-21:
Thanks for the heas up, @kroimon ! 
I removed the reference from the doc in the dev branch.

commit 69b495a0 (2021-08-10): Validate preset names and group names at compile time (fixes #15) (#47)

* Validate preset names and group names at compile time (fixes #15)

* Small fix for using a less confusing var name.