mmakaay/esphome-xiaomi_bslamp2 #25: Device won't power up

loefchen, 2021-05-17:
I compiled the firmware to my needs and managed to flash it, but it doesn't boot.

1. Compiled the firmware
2. Soldered the wires (at first I accidently connected the GPIO0 solder point to the ground spot beneath it, fixed that)
3. Flashed the firmware
4. Disconnected the flasher an power supply
5. Reconnected the power supply

The device won't boot. I tried a blank ESPhome instance, also a Tasmota32 binary. Could flash both of them, but the device does not boot any of the firmwares.

Do you have any idea, where the problem could be?

Kind regards, loefchen
[image]

erwinschoonhoven, 2021-05-17:
Have you tried flashing the original firmware back to the device? Does it
boot with the original firmware?

I had an issue with booting the device at first, but when I flashed the
original firmware back is worked again. I had this until I tried a
different flashing tool. Now it also works with the rewritten firmware.

loefchen, 2021-05-17:
I've used NodeMCU PyFlasher. I haven't backed up the original firmware (since I have another BSLamp2 in the worst case)...

Which flashing tool an which flasher did you use?

loefchen, 2021-05-17:
I solved it! It works usin ESPhome Flasher instead oder NodeMCU PyFlasher! Thanks a lot!

closed 2021-05-17 by loefchen

erwinschoonhoven, 2021-05-17:
Good to hear! Enjoy your new light.. :-)

mmakaay, 2021-05-17:
Ha, that went smooth :-)
Welcome to the club!

ESPHome Flasher is very useful, because it will not only flash the firmware.bin (which is the app that runs on the ESP), but also things like the bootloader and the initial OTA configuration (telling the bootloader that app nr 1 - there are two app spots for OTA support - must be started at boot).

You can use other flashing tools, but then you'd have to take care of these bits yourself.
It is describes in the manual for esptool, but writing that bit of documentation was a bit of a pain, having to find out how the pieces fit together ;-)

Therefore also my general advice: go for esphome-flasher for ESPHome projects, it's a breeze.

loefchen, 2021-05-18:
I just did not know that the Flasher Tool could be an issue. Life is learning ;)

Also, it's a cool thing that you made the custom firmware using ESPHome, I always wanted to try it out, but never dared using it and always used Tasmota for my Projects ;)

Thanks for your work, Maurice!

mmakaay, 2021-05-18:
You're most welcome :+1:

In fact, this project was my first real ESPHome work. I had done a few small projects before, which required no coding (a few RGB LED strips and a door bell). I figured that coding something for this lamp would teach me a lot more. An oh boy, was I right 😄