mmakaay/esphome-xiaomi_bslamp2 #27: Need binary file

maxmontgmx, 2021-05-21:
Hi Guys,
Would it be possible to share a firmware.bin file already complied with default configuration?
I don't have esphome environment to do that. And I don't have linux machin to install.
Thanks in advance for your support

mmakaay, 2021-05-21:
This is not really possible I'm afraid. The firmware also contains the configuration for the WiFi network. So you will have to compile the firmware with the correct configuration in it.

When compiling your own firmware, you have the additional advantage of being able to configure the lamp in any way you see fit. The default YAML config is pretty  good already, but when you want some other preset colors or behaviors, then you'll have to compile the firmware.

Are you going to use the lamp with Home Assistant?
If yes, then maybe you could install the ESPHome add on to get a compilation environment?

Currently, work is being done on a feature that allows configuration of the WiFi network via Bluetooth. When that is ready, distributing a basic firmware binary will be possible, since you can then setup the network manually on a running device. I plan  to release a binary firmware when that goes  into the stable ESPHome release.

maxmontgmx, 2021-05-21:
Thanks for this complete answer.
Maybe , hotspot configuration can make the config file be standard
I'm using Jeedom system, not HA. 
I'm just discorering ESPhome and HA. And I don't know if Jeedom manage ESPHome devices.

mmakaay, 2021-05-21:
If Jeedom can do MQTT or REST cslls, then it would be possible for sure! ESPHome native protocol API is only supported by Home Assistant, but use of ESPHome is no way limited to Home Assistant.

The captive portal might be an idea indeed. I will cook up a firmware with that tomorrow and try that on a Dev lamp. If that works, I will add the binary to the release.

mmakaay, 2021-05-21:
A few thought that came to mind:

One additional issue with a firmware like this, might be the naming of the device and entities.
The names are all coded into the firmware, so if one has two lamps, then the firmware would be a bit .. useless. 
I'll choose a sane default value for these  though, so people with only a single lamp can be helped.

What would be an issue for myself, is that upgrading the firmware would require opening the lamp again, to flash the device through the TX/RX on the board of the lamp directly. OTA is operational after the first flashing operation, but since users of a pre-compiled firmware probably won't be installing the esphome tooling, OTA flashing is not an option.

maxmontgmx, 2021-05-24:
Hi Maurice,
Thanks again for this grant support. I appreciate.
I know that jeedom supports MQTT but not in place in my setup. That requises a mosquito serveur. Not so easy.
Ok, if i go to this solution, i will build a linux VM for compilation.

mmakaay, 2021-05-24:
Sounds good. It really not that complicated. I hesitated quite a while before diving into ESPHome, but once I started, I was happily surprised by how quickly I was able to setup the software and compile firmware for my devices.

I'll close this issue then.

closed 2021-05-24 by mmakaay