mmakaay/esphome-xiaomi_bslamp2 #29: Remote control application

maxmontgmx, 2021-05-25:
Hi guy,
Sorry still some questions before entering in this new World.
After flashing ESPHome firmware, how can we directly control the lamp?
Which application?
And would it be possible to integrate the lamp in Google Home without HA?

Thanks again for your answer.

mmakaay, 2021-05-25:
ESPHome integrates with HA, that is the natural fit. If you don't run HA, this might not be the firmware for you.
Having said that: if you have some type of support for MQTT in your setup, then you could compile MQTT support into the firmware instead of the native ESPHome API, which would allow you to control the lamp using MQTT. Not particularly an easy quick job though, but it is possible for sure.

But when you say "Google Home", I wonder: isn't Google Home still supported for the lamp? That one doesn't need the custom firmware to work, right? Although there are some other benefits when using ESPHome, the main reason for the firmware's existence is to work around the loss of the LAN mode, which was needed for HA.

If I check for example [this page](https://www.yeelight.com/faqs/google-assistant), it looks like Google Home still might be a first class citizen for integration.

maxmontgmx, 2021-05-26:
I'm trying to find a work around on the LAN issue.
Until now, I'm using the lamp in local mode with Jeedom system and with Yeelight app for Google Home integration.
So 2 solutions, staying like that and I lose integration in Jeedom but I keep Google Home feature.
Or going to ESPHome to get back local LAN, but I loose Yeelight connection and Google Home integration.

If I go on ESPHome soution, I can try to connect google home though IFTTT.
I have to think about pro/cons for these 2 ways.

Thanks anyway

mmakaay, 2021-05-26:
A nice heterogenous setup there, that complicates things. That is in fact why I'm a fan of Home Assistant, which makes it possible to mix and match all these different systems. But that's not a switch that one would easily make I guess, when there's already a working system in place.

I'll close this ticket for now. I'd be interested to hear what your final outcome was though!

closed 2021-05-26 by mmakaay

maxmontgmx, 2021-05-26:
I see that some people played with CLI commands on serial console with Xiaomi Gateway to reopen the communication port.
I find an interesting document which enumerates these commands : https://dontvacuum.me/talks/DEFCON26-IoT-Village/DEFCON26-IoT-Village_How_to_Modify_Cortex_M_Firmware-Xiaomi.pdf
Maybe  the Local LAN mode can be enable via a command like psm-set to set a kind of environment variable.
And the command psm-dump may list the avialable variables to get the name.
It could be another way....