mmakaay/esphome-xiaomi_bslamp2 #32: Question: transition_length in mqtt message possible?

grasshide, 2021-05-31:
Hi @mmakaay 
Tanks for the incredible work you've done here!
One humble question. I use node-red to send mqtt command to my lamp. Is it possible to include the transition_length per request?
Example to set transition length to 4 seconds:
`msg.payload = {"state":"ON","color": {"r": rgb.r, "g": rgb.g, "b": rgb.b}, "brightness": 255, **"transition_length": 4000**};`
Thanks!

mmakaay, 2021-05-31:
I had to dive into the ESPHome core code myself, to see how the MQTT linked to the standard light command handling, but I think I found a solution for you. Can you try the following code?

```javascript
msg.payload = {"state":"ON","color": {"r": rgb.r, "g": rgb.g, "b": rgb.b}, "brightness": 255, "transition": "4s"};
```

grasshide, 2021-06-01:
> ```js
> "transition": "4s"
> ```

Thank you very much @mmakaay . That worked exactly as expected.
I was digging into your code instead of the esphome code and could not for the life of me figure that one out.
Thanks again!

closed 2021-06-01 by grasshide

mmakaay, 2021-06-01:
Great, cheers!