mmakaay/esphome-xiaomi_bslamp2 #33: Thanks and something offtopic

nob0dy80, 2021-06-04:
Hi Maurice,

first, thanks for your awesome work. Worked very good :)

I got  a other Xiaomi device (from their subcompany smartmi) that is more horrible as the bedside lamp.
https://www.indiegogo.com/projects/smartmi-p1-purifier-cleaner-freshen-air-in-10-mins/

It has also an ESP32 so it should be possible to "free" it like you did i with the bedside lamp.
My problem is, i cant do it alone cause of missing knowledge.
I got all tools that are needed (flashers, cables, ttls and courage :) )

Are you willing to give me some support to this ? Project could be published over your github page. Not interested in earning praises for this one....just love to tinker with it ;)

Greets from Germany

mmakaay, 2021-06-04:
You're welcome, glad you like it :-)

Reverse engineering can be quite a daunting task at times, depending on the components that you run into. With the lamp, I thought it would be an easy one, but it took me weeks of measuring, reverse engineering and coding to get everything working correctly (while I tought I'd be done it a few days at most ;-) )
If you want support and some more thinking heads for the P1, then maybe you could join the ESPHome Discord (I'm on there as well).
Note that having one device might be a bit hard to work with. I bought a separate lamp, so I could run my dev lamp and an unpatched lamp in parallel, giving me the opportunity to compare outputs and review what the original device was doing.

Anyway, step one would be to get the lamp open (in a  reversable manner ;-) ) and to describe the various parts that you find in there. See if the PCB(s) are easy to trace, if there are any debug pads for TX/RX/GND/GPIO at least, to make the flashing easy. That kind of stuff.

You can of course also use the Home Assistant community forums to post your progress on this, to see if you can find others that want to help out with ideas.

nob0dy80, 2021-06-04:
Hey Maurice,

thanks for you reply. Really appriciated.

i know that the reverse works are a thing for itself. but you get routined on that one too. analysing one firmware running on a esp32 makes you better in understanding the next one running on it.
so you were my first try asking for help.

Joining the Discord sounds like a good plan for the next step.

I got more than one device. the price was really neat.  and i'm not even mad if one device brakes. i guess thats part of that hobby,or? ;)

My goal, for sure, isnt a Home Assistant integration. It should be a integration for everyone. So one of my thoughts was going to a tasmota build. I dont know any smart home system that cant work with mqtt anyway. Myself is working with IoBroker. 

If it's cool for you, i'm going to open the P1 this weekend and deploy some photos to you. 
my first goal for sure is dumping the original firmware down for recovery works if anything wents wrong.

i dont have any problems to tear down devices and handle them propery, but when it comes to the software, reverse engineering part i'm out. And thats the part where i hope to get some help from you. :-)

Kind regards

nob0dy80, 2021-06-04:
joined discord, whats your name there? nothing starts with mma* ;)

mmakaay, 2021-06-04:
On Discord, I'm MauriceM, but there are other friendly people out there who might be willing to help (and are actually more versed with reverse engineering at the hardware level than me).

Tasmota is always an option, but why not try and go for ESPHome with MQTT?
I have done a bit of Tasmota, but I dropped it in favor of ESPHome when I found out about it.
Likely based on pure opinionated qualifications ;-)

mmakaay, 2021-06-04:
@nob0dy80 found the way to Discord, so closing this ticket.

closed 2021-06-04 by mmakaay