mmakaay/esphome-xiaomi_bslamp2 #34: [BUG] Won't compile with ESPHOME 1.19.1

Kaibob2, 2021-06-18:
**Describe the bug**
Won't compile with ESPHOME 1.19.1

**To Reproduce**
Steps to reproduce the behavior:
Upgrade ESPHOME from last version to 1.19.1
Try to install the freshly compiled firmware

**Expected behavior**
Compiles fine and uploads like it did before

**Additional context**
LOGFILE from ESPHOME compile:
```
INFO Reading configuration /config/esphome/yeelight.yaml...
INFO Generating C++ source...
/config/esphome/.esphome/external_components/d1d49d2c/components/xiaomi_bslamp2/__init__.py:68: RuntimeWarning: coroutine 'gpio_pin_expression' was never awaited
  yield from cg.gpio_pin_expression({ "number": number, "mode": mode });
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/opt/esphome/esphome/__main__.py", line 738, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 725, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 332, in command_run
    exit_code = write_cpp(config)
  File "/opt/esphome/esphome/__main__.py", line 144, in write_cpp
    generate_cpp_contents(config)
  File "/opt/esphome/esphome/__main__.py", line 156, in generate_cpp_contents
    CORE.flush_tasks()
  File "/opt/esphome/esphome/core/__init__.py", line 596, in flush_tasks
    self.event_loop.flush_tasks()
  File "/opt/esphome/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/opt/esphome/esphome/__main__.py", line 136, in wrapped
    await coro(conf)
  File "/opt/esphome/esphome/coroutine.py", line 79, in coro
    ret = yield from _flatten_generator(gen)
  File "/opt/esphome/esphome/coroutine.py", line 134, in _flatten_generator
    to_send = yield from _flatten_generator(val)
  File "/opt/esphome/esphome/coroutine.py", line 117, in _flatten_generator
    val = gen.send(to_send)
  File "/opt/esphome/esphome/coroutine.py", line 79, in coro
    ret = yield from _flatten_generator(gen)
  File "/opt/esphome/esphome/coroutine.py", line 134, in _flatten_generator
    to_send = yield from _flatten_generator(val)
  File "/opt/esphome/esphome/coroutine.py", line 117, in _flatten_generator
    val = gen.send(to_send)
  File "/opt/esphome/esphome/coroutine.py", line 79, in coro
    ret = yield from _flatten_generator(gen)
  File "/opt/esphome/esphome/coroutine.py", line 134, in _flatten_generator
    to_send = yield from _flatten_generator(val)
  File "/opt/esphome/esphome/coroutine.py", line 117, in _flatten_generator
    val = gen.send(to_send)
  File "/opt/esphome/esphome/coroutine.py", line 79, in coro
    ret = yield from _flatten_generator(gen)
  File "/opt/esphome/esphome/coroutine.py", line 117, in _flatten_generator
    val = gen.send(to_send)
  File "/config/esphome/.esphome/external_components/d1d49d2c/components/xiaomi_bslamp2/__init__.py", line 68, in make_gpio
    yield from cg.gpio_pin_expression({ "number": number, "mode": mode });
TypeError: cannot 'yield from' a coroutine object in a non-coroutine generator
```

Kaibob2, 2021-06-18:
I fixed it myself. Hopefully the commit get's merged (if correct)
See https://github.com/mmakaay/esphome-xiaomi_bslamp2/pull/35

mmakaay, 2021-06-19:
I already have a branch ready for compatibility with 1.19.0, so I will merge and release asap.
Life got in the way (i.e. a leaky roof and stuff resulting from that).
I'm planning to finish up some things this weekend, and this one's at nr 1 on my list.

PR #36 (2021-06-19): Compatiblity new coroutine
Some internal changes in ESPHome broke the compilation of this component.
This pull request contains the changes required to fix this.

Fixes #34
closed 2021-06-19 by mmakaay