mmakaay/esphome-xiaomi_bslamp2 #37: Unable to flash

cracrama, 2021-06-20:
Hello I'm having problem with flashing bedside lamp - 
I have copied config file - (i don't need to change anything) I have pasted it to esphome in HA. Later I managed to compile firmware (after error correction) - I have downloaded firmware and esphome-flasher - I connected  adapter and im getting "Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: Timed out waiting for packet header "
pins connection - rx-tx , tx-rx, gnd-gnd, gnd-gpio
its same story when i'm trying to update through server - (rasppi). i have tried different drivers and still nothing read that connecting gpio-gnd is not everything - you also need to press reset button? but where is reset button? my adapter doesn't have one - just holes with rts , cts.
I also tried different drivers but no sucess. 
Plese advise.

mmakaay, 2021-06-21:
There is no reset button on the lamp. When comparing this setup with for example a NodeMCU, then:
* the RESET button equivalent is "plug in the power"
* the FLASH button equivalent is "connect GPIO0 to GND"
So that's why the correct sequence for flashing is "connect GPIO0 to GND, then plug in the power".

The adapter is recognized by esphome-flasher? You can see its exposed COM-port in the drop down list?
The adapter has been used with other ESP32 projects for flashing?

cracrama, 2021-07-03:
Thank you for response and for amazing firmware - everything works fine now 

I will describe below what was the issue in my case (maybe that will help someone)

Errors I was getting: 

esphome HA
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2

esphome flasher
Using 'COM3' as serial port.
Connecting........_____....._____....._____....._____....._____....._____....._____
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: Timed out waiting for packet header

esphome flasher log:
view log
Using 'COM3' as serial port.
Unexpected error: could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)

Solution: Get yourself decent FTDI adapter!!! I know cheap ones on ebay  looks tempting but believe me - you will be using this tool many times in different projects so get one from reputable and trusted source. I went through 2 cheap adapters and when i finally got decent one everything went smooth as per author instruction. 

Also - check your soldering (it wasn't main reason however i found during the initial testing that not everything was connected correctly even when soldering was looking "fine" - resoldering pads helped. 

Thank you for great firmware and advice 
Best regards

closed 2021-07-03 by cracrama

mmakaay, 2021-07-05:
"Get yourself decent FTDI adapter!!!", best advice ever :-D
Thanks for reporting back with the detailed update.
Great that you got it to work eventually!