mmakaay/esphome-xiaomi_bslamp2 #40: [FEATURE]

Progaros, 2021-07-13:
Colors are displayed in Home Assistant, but color temperatures are just shown as white
[image]
both lights have a similar color temperature, but in HA the bedside lamp (left) is shown as white

It would be great to see color temperatures in Home Assistant

mmakaay, 2021-07-21:
I don't recognize these icons.
How exactly can I reproduce this within Home Assistant?

Progaros, 2021-07-22:
I gave the light entity of the bedside lamp this icon (mdi:alarm-light) 
The shape shouldn't matter

mmakaay, 2021-07-22:
Ah, I think I know what you mean now.
Looking at my HA GUI, my light icon represents the RGB color when using color mode, but for color temperature it is always a grey color in different brightnesses.

I really have no idea how that part is handled within HA exactly.
Maybe it has something to do with the new color mode support in the recent versions of HA.
This will take some investigation to see what component is responsible for making this work.

mmakaay, 2021-07-22:
I've looked a bit further into it, and my guess is that I can't really fix this.
It looks like an upstream issue between HA and ESPHome. 
Recently, HA introduced the explicit "light_mode" feature, to take away some guessing about the mode in which a light is configured. For ESPHome, there's still work to be done to get the support fully in there. Check out this pull request for example:

See: https://github.com/esphome/esphome/pull/2012

The reason that the icon shows up as grey, is that when switching to color temperature mode, the device stays at "hs" mode in HA (hue, saturation), and the color is set to rgb(255,255,255) a.k.a. "white". With a halfway brightness, that results in a grey icon.

I'll close this ticket for my firmware, because I can't really fix this from within my code.
I might have to do compatibility changes to integrate the new light mode support with my code base, but I will create a separate issue ticket if that indeed happens at some point.

closed 2021-07-22 by mmakaay