mmakaay/esphome-xiaomi_bslamp2 #42: [FEATURE] Implement light effects

megane999, 2021-08-04:
How to emplement candle effect using only white cold and warm led?

mmakaay, 2021-08-05:
It would require an effect to be written, which implements the candle-like effect.
Maybe somebody wrote one for another light already, which can be modified.
Otherwise it'd need to be done from scratch.

I can give this a try, but I haven't done much with effects myself.

megane999, 2021-08-05:
effects uses each led as light output, and when I want to create candle effect it uses all leds

mmakaay, 2021-08-05:
I am not sure if I fully understand what you mean there.
The LEDs in this lamp are not individually addressable. The only thing you can play with in an effect, is the color of the light (either in RGB or Color Temperature mode). It is not possible to make the lamp look like there is fire inside, by targeting individual LEDs. So in fact, the only thing you can do is "use all LEDs", but with the restriction "use all LEDs at the same time with the same color/brightness".

The only LEDs that are individually addressable since the latest firmware version, are the ones on the front panel. So one could write some form of effect that makes the front panel slider flicker like a candle for example. Not sure if that would look any good and if that would be what you are after here.

Do you have an example video of a lamp / this lamp doing the desired effect maybe?