mmakaay/esphome-xiaomi_bslamp2 #45: [BUG] Unable to flash using ESPHome 1.20.4

bengizmo, 2021-08-08:
**Describe the bug**
I'm unable to flash ESPHome firmware onto my xiaomi_bslamp2 using ESPHome 1.20.4  on hassio. See attached files for my config and the flash log. These errors look to be of interest:
> src/esphome/components/xiaomi_bslamp2/light/color_rgb_light.h:255:11: error: 'class esphome::light::LightColorValues' has no member named 'get_color_mode'
> src/esphome/components/xiaomi_bslamp2/light/color_rgb_light.h:255:38: error: 'esphome::light::ColorMode' has not been declared
> src/esphome/components/xiaomi_bslamp2/light/light_output.h:31:12: error: 'class esphome::light::LightTraits' has no member named 'set_supported_color_modes'

**To Reproduce**
Steps to reproduce the behavior:
1. Added the appropriate folder and files to my custom_components directory
2. created a new yaml file for the device in the ESPHome web configurator
3. Copied and modified the example yaml, and verified that there was no errors in the config
4. Attempted to flash with a FTDI chip connected to the Raspberry Pi 4 over USB
5. Encountered error messages and no flash :( 

**Expected behavior**
The device to successfully flash.

**Additional context**
Using a Raspberry Pi 4, Home Assistant 2021.8.4, ESPHome 1.20.4 via supervisor.

[xiaomi_bslamp2_esphome1.20.4_config_yaml_8August2021.txt](https://github.com/mmakaay/esphome-xiaomi_bslamp2/files/6951516/xiaomi_bslamp2_esphome1.20.4_config_yaml_8August2021.txt)
[xiaomi_bslamp2_esphome1.20.4_flashlog_8August2021.txt](https://github.com/mmakaay/esphome-xiaomi_bslamp2/files/6951517/xiaomi_bslamp2_esphome1.20.4_flashlog_8August2021.txt)
[hassio_config.txt](https://github.com/mmakaay/esphome-xiaomi_bslamp2/files/6951518/hassio_config.txt)

Kaibob2, 2021-08-09:
Same here.

NUC, Debian Buster, Home Assistant 2021.8.4, ESPHome 1.20.4,

mmakaay, 2021-08-09:
This is a known issue. The current codebase contains some compatibility updates for the most recent Home Assistant and the upcoming ESPhome v1.21.0. unfortunately, this code is not backward compatible.

To solve it for now, either:

* Use the dev branch of ESPhome or
* Use the external component definition from below to use the code that is compatible with ESPhome v1.20.4.

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/mmakaay/esphome-xiaomi_bslamp2
      ref:  v1.1.0
```
Tomorrow (bed time here) I will check if I can make the code compile correctly for either version of ESPhome.

Kaibob2, 2021-08-10:
I installed the dev Branch in parallel to the main in HA and only flashed the lamp via the dev Branch. Will now leave both instances running until 1.21.0 final is released. Thanks for clarification.
Great job btw 👍

mmakaay, 2021-08-10:
Fixed the issue in #46 
The main branch will now compile with both the current and the upcoming version of ESPHome.
For future changes, I will introduce an extra branch, to not push changes that conflict with the stable version of ESPHome.

closed 2021-08-10 by mmakaay

Kaibob2, 2021-08-10:
It compiles, yes, but color and temperature are still not available like explained here https://github.com/mmakaay/esphome-xiaomi_bslamp2/issues/43

Is this intended behavior?

mmakaay, 2021-08-10:
**I would not call it "intended behavior", but its unfortunately "a phase that we have to go through".**

The intended behavior is to have the HA GUI expose the lamp configuration with two tabs: one for RGB and one for Color Temperature selection, as was the csase up to Home Assistant 2021.7.4. However, the ESPHome integration in Home Assistant 2021.8.0 introduced some changes that broke this behavior in the GUI.

**What caused the problem?**

The main reason for the problem is a simple oversight: the ESPHome team didn't have any light components that used the specific light mode configuration of this lamp (interlocking RGB + Color Temperature). Backward compatibility for this combination was therefore forgotten. As I understood, it must even be concidered accidental that things worked correctly in earlier versions of Home Assistant :-)

**What was changed for the Bedside Lamp 2 component?**

With ESPHome v1.20.0 (the current stable release), it is not possible to convince Home Assistant to provide the intended behavior as described above. The backward compatbility code in Home Assistant 2021.8.0 misses the required options.

With ESPHome v1.21.0 (currently under development), it will be possible to tell Home Assistant 2021.8.0 exactly what color modes are supported by the lamp. The code fixes for #43 were done to implement this.

Unfortunately, those code changes were not backward compatible with ESPHome v1.20.0. This caused the failing compilation for which this issue was posted. What I fixed for this issue, is that te code base now compiles with both ESPHome v1.20.0 and v1.21.0. I switch dynamically between the old and the new way to tell Home Assistant what color modes are supported.

_As you found out: the code now compile successfully with ESPHome v1.20.0. However since the firmware cannot tell Home Assistant 2021.8.0 what color modes to use, you experience the missing RGB and Color Temperature tabs._

**How to fix this?**

Once ESPHome v1.21.0 is released, then upgrading ESPHome and building + flashing fresh firmware will fix the issues.

Until then, it is possible to use the dev-version of ESPHome to compile the firmware. The dev version already contains the required changes that will go into the stable v1.21.0 release. Multiple users and myself have tested this firmware and confirmed that it works as intended.

When you have to wait for v1.21.0 for some reason, then we're back to the start of this post: then it is "a phase that we have to go through".