mmakaay/esphome-xiaomi_bslamp2 #65: [BUG] Can't import the external component

shokwaav, 2021-11-11:
**Describe the bug**
A clear and concise description of what the bug is.

I'm getting an error with importing the component on validating the esphome firmware.

`ERROR Unable to import component xiaomi_bslamp2.light:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/d1d49d2c/components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.text_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/d1d49d2c/components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.output:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/d1d49d2c/components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
[... 230 lines cut ...]
              next: preset
      - timing:
          - ON for at least 0.6s
        then:
          - preset.activate:
              next: group

# This sensor component publishes touch events for the front panel slider.
# The published value represents the level at which the slider was touched.
# By default, values range from 0.01 to 1.00 (in 20 steps). This range can
# be modified using the "range_from" and "range_to" parameters.
sensor:
  # When the slider is touched, update the brightness.
  # Brightness 0.01 initiates the light night mode, which has already
  # been handled above (by holding the power button). Therefore, brightness
  # starts from 0.02 here, to not trigger night mode using the slider.
  - platform: xiaomi_bslamp2
    id: ${id_slider_level}
    range_from: 0.02
    on_value:
      then:
        - light.turn_on:
            id: ${id_light}
            brightness: !lambda return x;
`

shokwaav, 2021-11-11:
Oh never mind, seems I use the example.yaml from the dev version.

It seems the main branch hasn't been updated yet?

closed 2021-11-11 by shokwaav

mmakaay, 2021-11-11:
> It seems the main branch hasn't been updated yet?

The main branch was becoming a bit hard to maintain, because I often have new stable changes ready, while ESPHome has not yet moved forward. When I pushed new code to main, it would break things for users that use old ESPHome versions.
So instead, the main branch is now "dev", on which I always work against ESPHome "dev", Once I get to a stable release, I branch to a release branch, for example "release/2021.10.0" for the current stable release.

So in fact, the "main" branch could be considered stale now. I won't remove it yet though, to not bother people that are still using it.

shokwaav, 2021-11-11:
> So in fact, the "main" branch could be considered stale now. I won't remove it yet though, to not bother people that are still using it.

I see - I think I got it confused because I clicked the example.yaml in the main branch instead of following the link int he readme.

Would it be possible to change the name of the "main" branch example.yaml name so it would be less confusing?

mmakaay, 2021-11-11:
Not really, because "main" is what's referenced in the existing configs that make use of it.
What was changed however, is the default branch for the repository.
If you browse to https://github.com/mmakaay/esphome-xiaomi_bslamp2 , then you will see that `dev` is selected as the default branch. So via that route you should get the most recent data.

In time, I certainly want to get rid of the `main` branch completely, possibly by renaming it to `legacy` or so, but at this point I think it's too soon for that.