mmakaay/esphome-xiaomi_bslamp2 #66: [BUG] Lightblue light when starting in colortemp mode

Xitro01, 2021-11-18:
**Describe the bug**
Yesterday evening I started my two hourly adventure of flashing my bslamp2's.
I'm quite happy with the results, so great job on this custom firmware! It offers me a lot of customizability which I yet have to discover how everything works.

The only thing that bothers me slightly is when you are using colortemp mode and you turn the lamp on it will first flash lightblue for half a second and even longer when you hold the brightness adjuster.

**To Reproduce**
Steps to reproduce the behavior:
1. Set your lamp in colortemp mode
2. Turn it off
3. Turn it on

**Expected behavior**
When you turn on your lamp in colortemp mode you expect it to go into the last color straight away, if that's not possible then just make a small delay where the lamp turns on a little bit later.

mmakaay, 2021-11-18:
When I follow the reproduction steps, the lamp turns off and on without glitches.
This is the case when using the on/off button on the lamp and the on/off switch in Home Assistant. I also tried turning the light on by touching and holding the slider on the lamp. That also turned on the lamp without issues.

Since you mention that holding the brightness adjuster slows down things: do you have a high log level configured by any chance? That will result in many log messages, putting a burden on the normal operation of the lamp. That's the only  route that I can think of where holding the slider would have impact.

Xitro01, 2021-11-18:
> When I follow the reproduction steps, the lamp turns off and on without glitches. This is the case when using the on/off button on the lamp and the on/off switch in Home Assistant. I also tried [...]

https://user-images.githubusercontent.com/19844787/142441655-6e8301ac-3ac9-4783-b4fd-66148d3e8f51.mp4

og messages, putting a burden on the normal operation of the lamp. That's the only route that I can think of where holding the slider would have impact.

I am using nothing more than your example configuration, so I do not have high log level configured. I have it on both of my lamps. See the video attached: https://user-images.githubusercontent.com/19844787/142441752-a9f8218c-571b-4fd5-90eb-e552bb793147.mp4

mmakaay, 2021-11-18:
That does look off indeed. I can't explain what it's doing there though. I have three lamps here, two production one development. I tested all three and they do a normal transitioning from off to on in color temperature mode.
Can you check what happens when you set the transition time to 0s?

Xitro01, 2021-11-18:
> That does look off indeed. I can't explain what it's doing there though. I have three lamps here, two production one development. I tested all three and they do a normal transitioning from off to on in color temperature mode. Can you check what happens when you set the transition time to 0s?

Might sound a bit noobish, but I do not see transition time anywhere in the configuration files. How do you suggest to apply that?

mmakaay, 2021-11-18:
It's [documented in the `packages/core.yaml` file](https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/packages/core.yaml), but no problem if you've missed that. It doesn't make you a noob.
What you can do, is add an extra substitution to your YAML file:

```yaml
substitutions:
  default_transition_length: 0s
```

mmakaay, 2021-11-18:
Yes, that helps. I do see a slightly blue phase when turning on the light. I am not sure if I can fix that cleanly. The lamp is very picky in the low level brightnesses. 

Maybe I can skip part of the brightness increase, or maybe it's cleaner to start at a warmer tint for the low levels.

Xitro01, 2021-11-18:
Happy to test both! Let me know when you have the time, no rush.

mmakaay, 2021-11-20:
Sure thing. I now have seen it, so now I can't leave it as-is 😄 Got a few things in queue, but I'll definitely will work on a fix.

Xitro01, 2021-11-22:
> Sure thing. I now have seen it, so now I can't leave it as-is 😄 Got a few things in queue, but I'll definitely will work on a fix.

Haha, I understand. I also have a little bit of OCD where everything has to work perfectly.