mmakaay/esphome-xiaomi_bslamp2 #67: Support for Yeelight LED Smart LAMP D2 - YLCT01YL

manunited10, 2021-11-21:
Greetings, I was hoping maybe there could be a support for Yeelight  LED Smart LAMP D2 - YLCT01YL. 
The board `looks` similar to the one here, yet simpler.
It's micro processor is `ESP-WROOM-32D` (esp32).
I attached a few photos as well.
It has two touch controls whose controller is ChipON KF8TS2716.
I haven't even turned it on. I hope somebody could guide me before I return it
[image]
[image]
!

mmakaay, 2021-11-21:
That lamp is different from the Bedside Lamp 2. The touch controller is different for a start.
The PCB is white, but apart from that I cannot tell if it's indeed similar to the bslamp2.
The CPU is the same, so knowledge from this project could be useful.
When the color output channels use the same mad schema as the bslamp2, then that part could be reused. No way to tell if that is the case from these photographs.

I don't own this type of lamp myself, so I cannot do any investigations on this. I suspect that support for this lamp would best belong in a separate repository because of the differences.
If anyone reverse engineers this lamp and finds that the LEDs are driven using the same logic as the bslamp2, then let me know and we'll have to find a way to make the color handling a shared component.

I'll close this ticket now, since it's not really an issue, nor fulfillable feature request for this project.
Feel free to add more comments to this discussion though.

closed 2021-11-21 by mmakaay

mmakaay, 2023-03-22:
I purchased a D2 lamp to see if I can get a working firmware for it.
You can follow progress on the HA Community forums:
https://community.home-assistant.io/t/custom-firmware-esphome-xiaomi-bedside-lamp-d2/550880

trmendes, 2024-11-27:
Hi @mmakaay . I'm checking the link and the last post is from 3 months ago. Are you still working on it?

I have a D2 lamp and it has the original firmware. I would like to try your firmware in a hope to fix an issue with the lamp.

Thanks for your work :)