mmakaay/esphome-xiaomi_bslamp2 #68: Question: Using power button for HA calls

Xitro01, 2021-11-22:
So this time not a bug, also not really a feature request. Just a question if something is possible.

Would it be possible to light the power LED when an entity in HA has a certain state? Also would it be possible to long press the power button when it has this state to change an entity in HA?

I have a cool idea with the lamp, but not sure if it is possible to pull off.

Thanks!

mmakaay, 2021-11-22:
The LEDs of the front panel can be controlled individually, so it is possible to give the power button light its own functionality.
For example, on my bedroom lamp, I installed a microwave presence sensor, and I power up the power button LED for 3 seconds when the sensor is triggered.  This allows me to see the button in the dark.

Long press and a HA call are possible too. That's something I use too. Long press triggers a power off script, so I can turn off everything in the house when going to sleep.

Xitro01, 2021-11-22:
That sounds quite like what I would like to achieve. After a certain time (9PM) I would like to have the power button lid if my "House mode" is "Default". If I long press the button "House mode" should go to "Night mode".

Do you have some configuration examples for me, to achieve this? Would be great, thanks in advance.

mmakaay, 2021-11-22:
Sure thing, I'll share some stuff when I'm behind my pc.
I will put my config in a config package on the GitHub repo, so anybody can use it for inspiration.

mmakaay, 2021-11-22:
I have just gone through the upgrade procedure for the lamp configuration, so the config is now compatible with the latest releases of ESPHome and the bslamp2 external component. I have not yet created a configuration package yet. However, here's the configuration of my bedroom lamp for some inspiration. Note that I've got some local configuration packages for setting up WiFi for example, but I think you'll find what you need in here.

```yaml
substitutions:
  name: bedside-lamp-bedroom
  friendly_name: Bedside Lamp Bedroom
  default_transition_length: 500ms
  
  # For packages/base.yaml
  ip: !secret net_bedside_lamp_bedroom_ip

external_components:
  - source:
      type: git
      url: https://github.com/mmakaay/esphome-xiaomi_bslamp2
      ref: dev
    refresh: 60s
    
packages:
  my_base: !include packages/base.yaml
  my_fixed_ip: !include packages/fixed_ip.yaml
  bslamp2:
    url: https://github.com/mmakaay/esphome-xiaomi_bslamp2
    ref: dev
    files:
      - packages/core.yaml
    refresh: 0s

logger:
  level: DEBUG

api:
  reboot_timeout: 0s

globals:
  - id: current_brightness
    type: float
    initial_value: "0.0"
  - id: show_power_button
    type: bool
    initial_value: "false"
  - id: light_is_off
    type: bool
    initial_value: "true"

script:
  - id: update_front_panel
    mode: restart
    then:
      - front_panel.turn_off_leds: ALL
      - if:
          condition:
            or:
              - text_sensor.state:
                  id: my_light_mode
                  state: rgb
              - text_sensor.state:
                  id: my_light_mode
                  state: white
          then:
            - globals.set:
                id: light_is_off
                value: !lambda return false;
            - front_panel.turn_on_leds: [ POWER, COLOR ]
            - front_panel.set_level: !lambda return id(current_brightness);
      - if:
          condition:
            - text_sensor.state:
                id: my_light_mode
                state: night
[... 149 lines cut ...]
    on_state:
      then:
        - globals.set:
            id: show_power_button
            value: !lambda return x;
        - script.execute: update_front_panel

# This sensor component publishes touch events for the front panel slider.
# The published value represents the level at which the slider was touched.
# By default, values range from 0.01 to 1.00 (in 20 steps). This range can
# be modified using the "range_from" and "range_to" parameters.
sensor:
  # When the slider is touched, update the brightness.
  # Brightness 0.01 initiates the light night mode, which has already
  # been handled above (by holding the power button). Therefore, brightness
  # starts from 0.02 here, to not trigger night mode using the slider.
  - platform: xiaomi_bslamp2
    id: my_slider_level
    range_from: 0.02
    on_value:
      then:
        - light.turn_on:
            id: my_light
            brightness: !lambda return x;
```

mmakaay, 2022-01-05:
Can this issue be closed @Xitro01 ?
Or do you need more help with this?

Xitro01, 2022-01-05:
> 

Sure, you may close the issue. Haven't been able to get this to work yet, but don't have a lot of time to fiddle around.
I also found a completely other route to achieve what I would like.

mmakaay, 2022-01-05:
Okay, feel free to come back to this issue if and when you get back to this idea.

closed 2022-01-05 by mmakaay