mmakaay/esphome-xiaomi_bslamp2 #69: State saving doesn't work anymore.[BUG]

cracrama, 2021-11-25:
Hiya 
It's not maybe  an"issue" but i don't know exactly how to recreate previous behaviour. 

Issue description: 
  On previous config, after power outage, lamp was returning to it's previous state. (on, color, light)

**To Reproduce**
Steps to reproduce the behavior:

1. Disconnnect power supply
2. Reconnect
3. Lamp resetting to it's "default" settings. 

**Expected behavior**
I would like to keep state which lamp was before power disconnection. 

Please advise.

Kind regards.

Kind regards.

mmakaay, 2021-11-26:
That is not directly an issue with the component code, but a bug in ESPHome. You can follow progress on this in:
https://github.com/esphome/issues/issues/2682

I will close the issue for this project, since the issue is In the external code. Any extra findings you have, can be added to the ESPHome issue

cracrama, 2021-11-26:
Ok many thanks for quick response
Best regards and Many thanks for support.

mmakaay, 2021-11-26:
I'll reopen this issue, making it a TODO for myself.
I'm going to take a look to see if I can find the issue myself.
Would be nice to have this fixed for ESPHome 2021.12.0.

mmakaay, 2021-11-26:
In the [ESPHome issue](https://github.com/esphome/issues/issues/2682), I debugged the issue and it turns out that state saving works perfectly well, but restoring is what fails. I wrote a fix in [this pull request](https://github.com/esphome/esphome/pull/2805). Using this patch, my lamp restores its state after a reset.

When the PR  gets merged, I'll close this issue.

mmakaay, 2021-11-26:
The PR got merged, so the fix will be in the next version of ESPHome. Closing this issue.
Thanks for the extra push @cracrama 👍 I noticed the issue too before, but hadn't taken the time yet to actually search for the root cause. Others will be happy too, since the issue is generic for ESP32 devices.

closed 2021-11-26 by mmakaay