mmakaay/esphome-xiaomi_bslamp2 #70: [BUG] No serial data received.

joaovitoriasilva, 2021-12-23:
**Describe the bug**
Hello,

So when I try to flash the firmware I get "Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: No serial data received." error.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. In the app, select the COM port of your serial adapter.
2. Then select the firmware.bin file to flash onto the lamp.
3. Power up the lamp with GPIO0 connected to GND.
4. Click the "Flash ESP" button to flash the firmware.
5. I get the error and it is not flashed

**Additional context**
If I connect the lamp without entering flashing mode I can see the lamp logs like "looking for WiFi", "Turn on and off actions", etc, so I believe it is not a problem related with the serial connector/drivers/etc
Also on the ESPHome device YAML I only added the example provided with different SSID access and related

joaovitoriasilva, 2021-12-23:
Hello,

so it appears to be a soldering issue. I did it again and it is now working. 
two lamps flashed! Really cool! I am not at home right know so I don’t know if they will connect to the wifi, but everything seems OK. 

thank you for the awesome work.

closed 2021-12-23 by joaovitoriasilva

mmakaay, 2021-12-23:
Ah, you beat me to it. Indeed, if such issue happens, it's likely that there's a soldering issue.
In this case, the my guess would have been an issue with the RX pad on the lamp, since TX was functional.
Great you got it to work, enjoy the firmware!