mmakaay/esphome-xiaomi_bslamp2 #75: [BUG] Night mode bedside lamp2

ferrarii1, 2022-01-17:
**Describe the bug**
When I turn on the night mode on the lamp, it turns on the green backlight

**To Reproduce**
Steps to reproduce the behavior:
1. Pressing the off button 2 seconds
2. The lamp turns on a green backlight, as in the photo.

**Expected behavior**
I expect to see a red backlight

There is a second similar lamp, it shows red light as it should be. Firmware is the same. [bedside-lamp2-1.txt](https://github.com/mmakaay/esphome-xiaomi_bslamp2/files/7879325/bedside-lamp2-1.txt)

How can you fix the problem?

[image]
[image]
[image]

mmakaay, 2022-01-20:
This is kind of a duplicate for https://github.com/mmakaay/esphome-xiaomi_bslamp2/issues/74
The main issue here is hardware, not software. I have multiple lamps that all have different color shades when I turn on the night light mode. The LED circuitry doesn't produce really predictable colors at the night light output level. You'll have to find the RGB values that produce a shade that you like, and use those values in the config.

ferrarii1, 2022-01-21:
When I turn on the red color in night mode through HA, the lamp stops shining.

ferrarii1, 2022-01-24:
@mmakaay Maybe I need to change the frequency?

mmakaay, 2022-01-24:
The frequencies that I use have been reverse engineered by actually measuring all the ESP32 outputs, so I'm not sure if that would help. I wouldn't use other frequencies myself, simply because I don't exactly know how the LED driver circuitry would react to that. There's a chance that your hardware simply doesn't handle the red channel at the 1% brightness.

If you want to play with anything, it would be the code from https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/components/xiaomi_bslamp2/light/color_handler_night_light.h
That is the code that drives the GPIO's when the lamp is in night light mode (i.e. 1% brightness).

The only thing that I can think of to provide better support for night light tinkering, would be to make the RGB factors that are used in abovementioned code (`color_handler_night_light.h`) configurable through the YAML when tweaking them is needed to calibrate the outputs. That would make it possible to crank up the red channel a bit, to see if it kicks in.

ferrarii1, 2022-01-24:
Try to change the value in line 40 red = 0.968f; on red = 0.972f; ?

mmakaay, 2022-01-24:
That should be in the other direction. The higher the value, the more the LED channel is dimmed.

ferrarii1, 2022-01-25:
it worked. thank you very much.
[i fork you project](https://github.com/ferrarii1/esphome-xiaomi_bslamp2/blob/dev/components/xiaomi_bslamp2/light/color_handler_night_light.h)

andyboeh, 2022-09-28:
@mmakaay It would be great if you could make the RGB values in night light mode configurable in the YAML - I'd like to make the night light a bit brighter than the default and this seems the way to go.

mmakaay, 2022-09-29:
There is no linear relation between the led driving levels and the brightness of the lamp around those low night light output levels. So I'm not sure if you'd be able to go a little bit brighter. However, it's worth a try of course 👍

I'll get back to this.

mmakaay, 2022-09-29:
Reopened, since this was not fixed yet.

luke-barnett, 2023-01-04:
I had the same issue and was also able to resolve it by adjusting the frequency value for the red channel.

Xitro01, 2023-02-03:
> I had the same issue and was also able to resolve it by adjusting the frequency value for the red channel.

Could you elaborate this a bit? Explain how you did this and what your approach was?
I have 2 of these lamps which have a 2 different colors in night light mode, one is yellow and one is orange.

I tried what was explained here: https://github.com/mmakaay/esphome-xiaomi_bslamp2/issues/74
But this does not work if you always put the lamps on 1% through HA and not with the power button.

luke-barnett, 2023-02-03:
> Could you elaborate this a bit? Explain how you did this and what your approach was? I have 2 of these lamps which have a 2 different colors in night light mode, one is yellow and one is orange.
>
> I [...]

I forked the repo and made a couple of key changes.

* In the base config updated the repo reference to point to my fork (this took me a bit of head scratching to start with as none of my other changes were having an effect.
* Tweaked the frequency value for the red channel to boost it and be on when in night light mode which fixed up the night light colour.

From there I updated my esphome config to point to my repo directly for the effected light and have been happy since. You may choose to just skip to this step and use my repo fork or the same from @ferrarii1 but I'm making no guarantees to update it in future and will delete it when I no longer need it.

I went down this route after confirming that in night light mode if I picked a red the light wouldn't turn on. If you are getting different colours from the green we had you'll probably find you need to tweak a different colour channel.

mmakaay, 2023-02-04:
As soon as a change that I requested (and which was accepted and implemented) for ESP-IDF makes its way into the release packages for ESP-IDF, I will schedule a weekend of happy hacking on the device firmware and get a few lined up features done. This will be one of them: being able to calibrate the night light from the YAML config.