mmakaay/esphome-xiaomi_bslamp2 #78: newbie questions

passuff, 2022-03-27:
First of all thanks for this nice project. 
I have two bedside lamps. One is working with LAN access V2 30 and the other one had a firmware update to V2 41 and thus LAN access was closed. 
I'm using iobroker and I was thinking I could use this firmware to activate MQTT to make the lamp work in iobroker again.
I was following the guide and have several questions:

I compiled the FW with example.yaml (edited for wifi credentials) and started the lamp. I  checked the router and it sees the lamp got an ip. And then? how can I access the lamp and add the MQTT credentials/settings if I don't have home assistant? seems like there is no web interface, right?

Because I was not able to do anything smart with it but just using the buttons I tried to backup the FW (V30) from the working lamp and write it to the broken lamp. Unfortunately it was not working. Seems like the lamp is now in some boot/reset loop. After plugin the power, the lamp blinkt 7 times and then goes off and starts again.
Buttons are working, network is not (no IP 

I checked the V30 bin:

`ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿé ”‘@î               ÿ?   ÿÿÿÿ ÿ?d   	þ¬          €       À   àPRO APP boot [0;33mW (%d) %s: WDT reset info: %s CPU PC=0x%x (waiti mode)[0m
 [0;33mW (%d) %s: WDT reset info: %s CPU PC=0x%x[0m
  is not bootable [0;31mE (%d) %s: Factory app partition%s[0m
 [0;31mE (%d) %s: Factory test app partition%s[0m
 [0;31mE (%d) %s: OTA app partition slot %d%s[0m
 [0;32mI (%d) %s: Loaded app from partition at offset 0x%x[0m
 RF data test app WiFi data Unknown data OTA data Unknown app OTA app factory app unknown [0;31mE (%d) %s: bootloader_mmap(0x%x, 0x%x) failed[0m
 [0;31mE (%d) %s: Failed to verify partition table[0m
 [0;32mI (%d) %s: Partition Table:[0m
 [0;32mI (%d) %s: ## Label            Usage          Type ST Offset   Length[0m
 [0;32mI (%d) %s: %2d %-16s %-16s %02x %02x %08x %08x[0m
 [0;32mI (%d) %s: End of partition table[0m
 QIO QOUT DIO DOUT FAST READ SLOW READ 2MB [0;32mI (%d) %s: Detected ESP32D2WD[0m
 [0;32mI (%d) %s: Detected ESP32PICOD2[0m
 [0;32mI (%d) %s: Detected ESP32PICOD4[0m
 [0;32mI (%d) %s: Detected ESP32[0m
 [0;33mW (%d) %s: PRO CPU has been reset by WDT.[0m
 [0;33mW (%d) %s: APP CPU has been reset by WDT.[0m
 854b5cd [0;32mI (%d) %s: ESP-IDF %s 2nd stage bootloader[0m
 [0;32mI (%d) %s: compile time 04:41:35[0m
 [0;32mI (%d) %s: Enabling RNG early entropy source...[0m
 [0;32mI (%d) %s: SPI Flash RID  : 0x%X[0m
 [0;32mI (%d) %s: SPI Flash  MF  : 0x%02X[0m
 [0;32mI (%d) %s: SPI Flash  ID  : 0x%04X[0m
 [0;31mE (%d) %s: failed to load bootloader header![0m
 [0;32mI (%d) %s: SPI Speed      : %s[0m
 [0;32mI (%d) %s: SPI Mode       : %s[0m
 [0;32mI (%d) %s: SPI Flash Size : %s[0m
 [0;31mE (%d) %s: load partition table error![0m
 [0;31mE (%d) %s: GPIO_25 is pull down ,get in test mode[0m
 [0;31mE (%d) %s: ota_info partition size %d is too small (minimum %d bytes)[0m
 [0;32mI (%d) %s: Defaulting to factory image[0m
 [0;32mI (%d) %s: No factory image, trying OTA 0[0m
 [0;31mE (%d) %s: ota data partition invalid, falling back to factory[0m
 [0;31mE (%d) %s: ota data partition invalid and no factory, will try all partitions[0m
 [0;33mW (%d) %s: Falling back to test app as only bootable partition[0m
 [0;31mE (%d) %s: No bootable app partitions in the partition table[0m
 [0;32mI (%d) %s: Disabling RNG early entropy source...[0m
 DROM [0;31mE (%d) %s: Image contains multiple %s segments. Only the last one will be mapped.[0m
 IROM [0;31mE (%d) %s: Assert failed in %s, %s:%d (%s)[0m
 1MB 4MB 8MB 16MB 40MHz 26.7MHz 20MHz 80MHz  ¿	ÿ?yÿ?Ã	ÿ?Ç	ÿ?Ë	ÿ?Ð	ÿ?Ö	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?Þ	ÿ?ä	ÿ?bootloader_flash [0;31mE (%d) %s: tried to bootloader_mmap twice[0m
 [0;31mE (%d) %s: bootloader_mmap excess size %x[0m
 [0;31mE (%d) %s: cache_flash_mmu_set failed: %d
[0m
 [0;31mE (%d) %s: bootloader_flash_read src_addr 0x%x not 4-byte aligned[0m
 [0;31mE (%d) %s: bootloader_flash_read size 0x%x not 4-byte aligned[0m
 [0;31mE (%d) %s: bootloader_flash_read dest 0x%x not 4-byte aligned[0m
 [0;31mE (%d) %s: bootloader_flash_write dest_addr 0x%x not %d-byte aligned[0m
 [0;31mE (%d) %s: bootloader_flash_write size 0x%x not %d-byte aligned[0m
 [0;31mE (%d) %s: bootloader_flash_write src 0x%x not 4 byte aligned[0m
[... cut ...] yÿ?  ñ?âú? 0ô? õ?}ÿ?`ô?ÿÿÿ   ÿóÿÿ   §ÿ?Óÿ?ÿÿ?dô?    hô?Tô?Xô?\ô? ô? ÿ?ÿÏÿÿÿ÷ÿÿÅ³¢‘%ÿ?[ÿ?@ð?‘ÿ?™ÿ?Ðÿ?Œ€ô?Hðõ?ÿ¿ÿÿýÿ?pâú?  ô?ÿÿÿßÿÿÿ¿   €ÿÿÿ   p  ÿÿ      ÿÿÿÿ÷   9ÿ?fÿ?•ÿ?Äÿ?
ÿ?ýÿ?0ô?(ÿ?ð	ÿ?Sÿ?ê	ÿ?ÿÿ     ~ÿ?$ô?   „„ô?   ±ÿ?ÿ  ïÿ?Aÿ?tÿ?ªÿ?õÿ?Oÿ?šÿ?ãÿ?  ó¿ÿÿ2  	ÿ?%	ÿ?  ÀÀÿÿ? ƒ	ÿ?D ð?\ ð?X† @Ÿ @Üž @Ð @(} @Ì @Ô @LÄ @œh @¸š @82@š @„š @$› @4… @ˆ› @ÈÂ @à• @$ ÿ?     ÿ?  ÿ?ÿßÿÿ¤• @  ÿ?D
ÿ?U`

It does actually not look like a fw but more like a log.

passuff, 2022-03-27:
This is what I got from the serial output while in the boot loop:

`
_|      _|  _|_|_|  _|_|_|    _|_|
_|_|  _|_|    _|      _|    _|    _|
_|  _|  _|    _|      _|    _|    _|
_|      _|    _|      _|    _|    _|
_|      _|  _|_|_|  _|_|_|    _|_|
JENKINS BUILD NUMBER: N/A
BUILD TIME: Apr 24 2020,14:51:08
BUILT BY: N/A
MIIO APP VER: 2.0.6_0033
MIIO MCU VER:
MIIO DID: 472165774
MIIO WIFI MAC: ec4d3e565384
MIIO MODEL: yeelink.light.bslamp2
ARCH TYPE: esp32,0x0000a601
ARCH VER: 637cf05
08:00:00.670 [I] miio_wifi: cc=CN, schan=1, nchan=13, max_tx_dbm=0
E (14638) mfi_i2c: Read data from slave fail -1.
read MfiCpInfo failed.add[0x20],regadd[0x00],len[0x8]
08:00:07.210 [E] no MFi co-processor is detected!
 (miio_net_init,1497)
Keystore initialised
Accessory is not Paired with any controller
Database initialised. Accessory Device ID: F2:82:C3:1B:47:6D
mDNS initialised
HAP Initialization succeeded. Version : 2.3.r1-5e8ddaf
Setup ID: CPAO
W (14718) wifi: ap start fail

W (14718) wifi: ap start fail

08:00:07.300 [I] did=472165774 hostname=MiBedsideLamp2-5384

E (14728) i2c: i2c driver install error
E (14728) mfi_i2c: I2C driver initial fail -1
I2c Initialize failed
Coprocessor Initialize failed
Accessory MFI Auth Initival fail[-1]

E (14748) mfi_i2c: Write data to slave fail -1.
read errCode failed.add[0x20],regadd[0x10],len[0x1]
E (14758) YGLC_HAP: Auth signature -1, alert 10s[cnt=33]
08:00:07.330 [I] new_bright=33 nl_mode=0

08:00:07.340 [I] sw_duty=-1 light_hints=0 hints_duty=5

08:00:11.220 [I] miio_instance:  miio_reboot_hook_default called
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5220
load:0x40078000,len:0
load:0x40078000,len:12348
entry 0x40079194
I (66) boot: Detected ESP32
I (32) boot: ESP-IDF 854b5cd 2nd stage bootloader
I (32) boot: compile time 04:41:35
I (32) boot: Enabling RNG early entropy source...
I (37) boot: SPI Flash RID  : 0xB20B4
I (41) boot: SPI Flash  MF  : 0xB4
I (45) boot: SPI Flash  ID  : 0x200B
I (49) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : DIO
I (58) boot: SPI Flash Size : 4MB
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (87) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 miio_fw1         OTA app          00 10 00010000 001e0000
I (102) boot:  4 miio_fw2         OTA app          00 11 001f0000 001e0000
I (110) boot:  5 test             test app         00 20 003d0000 00013000
I (117) boot:  6 mfi_p            Unknown data     01 82 003e3000 00001000
I (125) boot:  7 factory_nvs      WiFi data        01 02 003e4000 00004000
I (133) boot:  8 coredump         Unknown data     01 03 003e8000 00010000
I (140) boot:  9 minvs            Unknown data     01 fe 003f8000 00004000
I (148) boot: End of partition table
I (152) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3c260 (246368) map
I (212) esp_image: segment 1: paddr=0x0004c288 vaddr=0x3ffbdb60 size=0x03d88 ( 15752) load
I (215) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0x120564 (1181028) map
I (461) esp_image: segment 3: paddr=0x00170584 vaddr=0x3ffc18e8 size=0x00450 (  1104) load
I (461) esp_image: segment 4: paddr=0x001709dc vaddr=0x3ffc1d38 size=0x00580 (  1408) load
I (468) esp_image: segment 5: paddr=0x00170f64 vaddr=0x40080000 size=0x00400 (  1024) load
I (476) esp_image: segment 6: paddr=0x0017136c vaddr=0x40080400 size=0x1bd98 (114072) load
I (531) boot: Loaded app from partition at offset 0x10000
I (531) boot: Disabling RNG early entropy source...
08:00:00.030 [I] use sub model = 0

08:00:00.050 [I] Dysis standby IO=0

08:00:00.050 [I] light hints after turnoff, setting: 0 5

08:00:00.050 [I] initialize mode data:

08:00:00.050 [I] ModeData: 1 2700 3400 6500 0 0 0 0

08:00:00.060 [I] parse mode data done. ret=0

08:00:00.060 [I] ModeData: 2 16734047 16756806 9170548 57855 13395711 0 0

08:00:00.070 [I] parse mode data done. ret=0

08:00:00.070 [I] dysis mode idx init, cur_yglc_mode=2

08:00:00.080 [I] cur_ct=3400_0xd48 idx=2

08:00:00.090 [I] indicator delayoff setting: 0 15

08:00:00.090 [I] Dysis standby IO=1

08:00:00.120 [I] new_bright=100 nl_mode=0

08:00:00.120 [I] sw_duty=-1 light_hints=0 hints_duty=5

08:00:00.120 [I] touch version === 28

08:00:00.170 [I] miio_ot: httpdns enabled
08:00:00.170 [I] miio_ot: dlg enabled

_|      _|  _|_|_|  _|_|_|    _|_|
_|_|  _|_|    _|      _|    _|    _|
_|  _|  _|    _|      _|    _|    _|
_|      _|    _|      _|    _|    _|
_|      _|  _|_|_|  _|_|_|    _|_|
JENKINS BUILD NUMBER: N/A
BUILD TIME: Apr 24 2020,14:51:08
BUILT BY: N/A
MIIO APP VER: 2.0.6_0033
MIIO MCU VER:
MIIO DID: 472165774
MIIO WIFI MAC: ec4d3e565384
MIIO MODEL: yeelink.light.bslamp2
ARCH TYPE: esp32,0x0000a601
ARCH VER: 637cf05
08:00:00.690 [I] miio_wifi: cc=CN, schan=1, nchan=13, max_tx_dbm=0
`

mmakaay, 2022-04-22:
The latest ESPHome firmware is based on the ESP-IDF framework and not the Arduino framework. The reason for this is that it solves *a lot* of issues with getting the firmware to work on the less-than-standard-ESP32 chip on the device. Downside of this is that MQTT and web server are not available for the time being, because the ESPHome ESP-IDF implementation does not (yet) have support for this built-in. Once that is taken care of, the lamp will be usable over MQTT again.

 Why the backed up firmware doesn't work, I can't tell. There are many weird error messages in the output. The firmware dump also contains stuff like WiFi and device setup, so maybe there's some conflict going on there, because the backup is not the one from the device itself?

mmakaay, 2022-04-25:
Ah, looking at the release notes of the latest stable ESPHome, I see that [a PR for adding MQTT to ESP-IDF builds](https://github.com/esphome/esphome/pull/2930) has been merged. So this might allow the use of MQTT with the Bedside Lamp 2. I'll see if I can test drive that somewhere this week.

Ice81, 2022-04-26:
Hi @all
I have been able to install MQTT on the lamp. Fhem has also received data:

IODev | MQTT2_FHEM_Server | 2022-04-26 16:46:07-- | -- | --
config_avty_t | bedside-lamp/status | 2022-04-26 18:00:28
config_brightness | true | 2022-04-26 18:00:28
config_clrm | true | 2022-04-26 18:00:28
config_cmd_t | bedside-lamp/light/bedside_lamp_rgbww_light/command | 2022-04-26 18:00:28
config_dev_ids | 50ec508a3e4d | 2022-04-26 18:00:28
config_dev_mdl | esp32doit-devkit-v1 | 2022-04-26 18:00:28
config_dev_mf | espressif | 2022-04-26 18:00:28
config_dev_name | bedside-lamp | 2022-04-26 18:00:28
config_dev_sw | esphome v2022.4.0 Apr 26 2022, 17:56:11 | 2022-04-26 18:00:28
config_effect | true | 2022-04-26 18:00:28
config_fx_list_1 | Slow Random | 2022-04-26 18:00:28
config_fx_list_2 | Fast Random | 2022-04-26 18:00:28
config_fx_list_3 | None | 2022-04-26 18:00:28
config_name | Bedside Lamp Light Mode | 2022-04-26 18:00:28
config_schema | json | 2022-04-26 18:00:28
config_stat_t | bedside-lamp/sensor/bedside_lamp_light_mode/state | 2022-04-26 18:00:28
config_supported_color_modes_1 | color_temp | 2022-04-26 18:00:28
config_supported_color_modes_2 | rgb | 2022-04-26 18:00:28
config_uniq_id | ESPsensorbedside_lamp_light_mode | 2022-04-26 18:00:28
debug | [0;33m[W][api.connection:083]: ESPHome Logs 2022.4.0 (::FFFF:192.168.178.65): Connection closed[0m | 2022-04-26 19:21:37
state | off | 2022-04-26 19:35:29
state_brightness | 5 | 2022-04-26 19:35:29
state_color_b | 255 | 2022-04-26 19:35:29
state_color_g | 42 | 2022-04-26 19:35:29
state_color_mode | rgb | 2022-04-26 19:35:29
state_color_r | 58 | 2022-04-26 19:35:29
state_effect | None | 2022-04-26 19:35:29
state_state | OFF | 2022-04-26 19:35:29
status | online | 2022-04-26 18:00:28

Readingslist were also created:
bedside_lamp_50ec508a3e4d:homeassistant/light/bedside-lamp/bedside_lamp_rgbww_light/config:.* { json2nameValue($EVENT, 'config_', $JSONMAP) }
bedside_lamp_50ec508a3e4d:bedside-lamp/light/bedside_lamp_rgbww_light/state:.* { json2nameValue($EVENT, 'state_', $JSONMAP) }
bedside_lamp_50ec508a3e4d:homeassistant/sensor/bedside-lamp/bedside_lamp_light_mode/config:.* { json2nameValue($EVENT, 'config_', $JSONMAP) }
bedside_lamp_50ec508a3e4d:bedside-lamp/sensor/bedside_lamp_light_mode/state:.* state
bedside_lamp_50ec508a3e4d:bedside-lamp/status:.* status
bedside_lamp_50ec508a3e4d:bedside-lamp/debug:.* debug

But unfortunately I can't manage to switch the lamp on or off via MQTT.

mmakaay, 2022-04-26:
I flashed my dev lamp with MQTT enabled, and what I see is that Home Assistant does not seem to understand the most important feature of the lamp: the actual lighting:

[image]

I do see a reboot switch that I added, which works perfectly.
And I to get reports on the current light status. When switching manually on the lamp, it toggles for example from 'rgb' to 'white'.
But the light option is not available in Home Assistant, so that matches your experience here.

I've never done much with the actual internals of ESPHome MQTT, so I don't see what the issue is at a glance. I will go and play with it. See if I can convince Home Assistant to see the lamp through MQTT.

mmakaay, 2022-04-26:
When inspecting the MQTT traffic, I see the following auto-discovery info:

```json
{
  "schema": "json",
  "clrm": true,
  "supported_color_modes": [
    "color_temp",
    "rgb"
  ],
  "brightness": true,
  "effect": true,
  "fx_list": [
    "Slow Random",
    "Fast Random",
    "Pulse",
    "None"
  ],
  "name": "Bedside Lamp Office RGBWW Light",
  "stat_t": "bedside-lamp-office/light/bedside_lamp_office_rgbww_light/state",
  "cmd_t": null,
  "avty_t": null
}
```

That looks about right, except for the missing `cmd_t`, which ought to contain the /command URL IIRC.
So there's a loose hanging thread to play with.

mmakaay, 2022-04-28:
I noticed quite a few of these messages during the build. This might possibly be related to the issue:
```
In file included from src/esphome/components/mqtt/mqtt_client.cpp:1:
src/esphome/components/mqtt/mqtt_client.h:280:3: warning: missing initializer for member 'esphome::mqtt::MQTTDiscoveryInfo::object_id_generator' [-Wmissing-field-initializers]
```

Ice81, 2022-04-28:
I did not get this problem.
I have attached my log file
[logs_bedside-lamp_upload.txt](https://github.com/mmakaay/esphome-xiaomi_bslamp2/files/8585503/logs_bedside-lamp_upload.txt)

Ice81, 2022-04-28:
I don't know why, but I don't get an MQTT button in the device overview. 
[image]

mmakaay, 2022-04-28:
That is to be expected I think. The light settings and light mode sensor are the elements that ought to be available throught mqtt autodiscovery. 

What functionality would you expect for an mqtt button?

mmakaay, 2022-04-30:
Ah wait, you are not looking at the mqtt integration, but the ESPHome integration in your screenshot. To see mqtt stuff, go to integrations and then mqtt in Home Assistant. The device should've available there as well (but likely in a broken state like on my setup).

Ice81, 2022-04-30:
After restarting my VM I now also see the MQTT Info button.

mmakaay, 2022-06-13:
I assume this issue can be closed?

closed 2022-10-09 by mmakaay