mmakaay/esphome-xiaomi_bslamp2 #79: [BUG] Unable to update firmware on new versions of ESPhome (2022.2.1)

cracrama, 2022-03-29:
Hiya me again 
From some time when I'm trying to update firmware of the lamp with new version of ESP home I'm receiven below mesage: 

--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
*** [/data/bedside-lamp/.pioenvs/bedside-lamp/src/esphome/components/api/util.o] Source `src/esphome/components/api/util.cpp' not found, needed by target `/data/bedside-lamp/.pioenvs/bedside-lamp/src/esphome/components/api/util.o'.
========================== [FAILED] Took 2.75 seconds ==========================

(Seems like some file is missing?)

**To Reproduce**
Just try to update firmware on new version of esphome ver; 2022.3.1

**Expected behavior**
Able to update firmware normally

Please advise
best regards. .

davidepanato, 2022-03-30:
I had the same problem, clean the build cache and it will work again.

cracrama, 2022-03-30:
Problem solved 
What a shame -  I missed that. 
Many thanks for help [davidepanato]:)
I'm closing issue. 
Best regards.

closed 2022-03-30 by cracrama