mmakaay/esphome-xiaomi_bslamp2 #80: [BUG] Noticable delay when turning the lamp ON/OFF from Home Assistant

ihrapsa, 2022-04-05:
**Describe the bug**
I've noticed that the lamp reacts quite slow on commands. It takes around 2s for a turn on or turn off command to be seen. I have other ESPhome devices in the house and they execute the command instantly.

**To Reproduce**
Steps to reproduce the behavior:
1. Toggle the lamp
2.  Check how long it takes for the lamp to turn on/off

**Expected behavior**
The lamp should be toggled instantly

mmakaay, 2022-04-22:
First off: apologies for responding so late. I missed the notification mail :-(

That doesn't sound like something that I can fix in code. I have three of these lamps and the interaction between de lamp and Home Assistant are virutally instant.

Do you have a high log level configured maybe? When logging a lot, the lamp will definitely slow down. I use logger `level: WARN` for my production lamps.

Another reason could be bad WiFi reception. Maybe you can check if placing the device somewhere else changes the behavior. If yes, then you know that the WiFI reception at the current spot is flaky.

ihrapsa, 2022-04-23:
@mmakaay Hi, thanks for the reply!

I've used the `examle.yaml` to build the binary and I can see now that in `core.yaml` logger level is set to DEBUG. I think this is causing issues since the lamp is <2m away from an AP.  Will have to test. However, it's still a real pain to update the lamp since I'm using ESPhome on arm (rpi4) and the espressif haven't fully added support for `aarch64` (they say v4.4 added support for aarch64 but I still can't manage to build the binary in esphome). Any thoughts/news on these that you might be aware of?

Thanks again!

mmakaay, 2022-04-23:
Hmm, I'll set the default log level to `WARN` then. The `DEBUG` must have slipped through somehow.

I don't know about the exact status for aarch64 unfortunately.
I haven't run into that issue myself and I have no test environment for it. I hope that it will progress quickly, since 64 bit is now officially supported by Raspberry.

ihrapsa, 2022-04-23:
Will get back and close the issue as soon as I test the firmware with logger on `WARN` if that's ok.

mmakaay, 2022-04-23:
Sure thing! Curious to know if that was it.

ihrapsa, 2022-04-28:
quick update:
The issue fixed itself somehow. I unplugged the lamp for about a week while I was away and now it responds instantly (the lamp had never been left unplugged that long before). I did nothing else as I can't remotely update the firmware. I did not update ESPHome either. What might have been? I'm a bit confused. 🤔

mmakaay, 2022-04-28:
Do you have multiple access points? Then it might have been connected to a less optimal one. I can't think of another

closed 2022-04-28 by mmakaay

ihrapsa, 2022-04-28:
> Do you have multiple access points? Then it might have been connected to a less optimal one. I can't think of another scenario as the moment.

I have two but they have different SSIDs. Will monitor the behavior from now on.
Thanks for all the help!

mmakaay, 2022-04-28:
Must have been solar flares then :-D
Let's hope things will remain stable now.