mmakaay/esphome-xiaomi_bslamp2 #81: [BUG] upload error

Inakhosz7, 2022-04-12:
reloading an existing configuration fails

INFO Reading configuration /config/esphome/bedside-lamp-right.yaml...
INFO Updating https://github.com/mmakaay/esphome-xiaomi_bslamp2@dev
INFO Updating https://github.com/mmakaay/esphome-xiaomi_bslamp2@dev
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing bedside-lamp-right (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
*** [/data/bedside-lamp-right/.pioenvs/bedside-lamp-right/src/esphome/components/api/util.o] Source `src/esphome/components/api/util.cpp' not found, needed by target `/data/bedside-lamp-right/.pioenvs/bedside-lamp-right/src/esphome/components/api/util.o'.
========================= [FAILED] Took 11.33 seconds =========================

mmakaay, 2022-04-22:
Please, when posting log output, post it as a code blob.
\`\`\`
use three backticks around the code
like this to make it readable
\`\`\`

I don't see the same errors and can generate and upload a new firmware without issues. Can you please specify your exact setup and versions of software that you are using?

Inakhosz7, 2022-04-27:
To compile and upload the firmware I am using the ESPHome application by Nabu Casa v2022.3.1.

mmakaay, 2022-04-28:
You mean that you are using it from the ESPHome add-on in Home Assistant?

First of all, make sure that you are using the latest version of ESPHome. That is what I cuse to compile.

mmakaay, 2022-06-13:
Any progress on this issue, or can I close it?

closed 2022-06-22 by mmakaay