mmakaay/esphome-xiaomi_bslamp2 #82: After firmware - nothing works

ADMiNZ, 2022-05-16:
Hello.

The firmware is assembled without errors.
Stitched without errors.

The download field does not work, the power buttons do not work and I do not see the device on the network :(

**xiaomi-bslamp2.yaml**

```
# --------------------------------------------------------------------------
# Substitutions
#
# These are substitutions as used by the configuration packages from below.
# You can uncomment and update the ones that you want to modify.
# --------------------------------------------------------------------------

substitutions:
  name: bedside-lamp2
  friendly_name: Bedside Lamp 2
  #light_name: ${friendly_name} RGBWW Light
  #light_mode_text_sensor_name: ${friendly_name} Light Mode
  #default_transition_length: 800ms

# --------------------------------------------------------------------------
# Load configuration packages
#
# These provide a convenient way to compose your device configuration from
# some functional building blocks. Pick and mix the blocks that you need.
#
# For customization you can override options in your config or you can
# copy the contents of these packages directly in your config file as
# an example for your own customizations.
#
# Available packages are:
# - core.yaml                : core components & hardware setup
# - behavior_default.yaml    : default device behavior
# - ota_feedback.yaml        : enable visual feedback during OTA updates
# - activate_preset_svc.yaml : 'activate_preset' service for Home Assistant
# --------------------------------------------------------------------------

packages:
  bslamp2:
    url: https://github.com/mmakaay/esphome-xiaomi_bslamp2
    ref: dev
    files:
      - packages/core.yaml
      - packages/behavior_default.yaml
      - packages/ota_feedback.yaml
      - packages/activate_preset_svc.yaml
    refresh: 0s

# --------------------------------------------------------------------------
# Use your own preferences for these components.
# --------------------------------------------------------------------------

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
  password: !secret api_password

ota:
  password: !secret ota_password
```

**Validate xiaomi-bslamp2.yaml**
```
INFO Reading configuration /config/esphome/xiaomi-bslamp2.yaml...
[... 402 lines cut ...]
```

**CMD**
```
C:\Users\*********\esptool>python.exe .\esptool.py --chip esp32 --port COM4 --baud 115200 write_flash 0x1000 bedside-lamp2.bin
esptool.py v4.0-dev
Serial port COM4
Connecting..................
Chip is unknown ESP32 (revision 1)
Features: WiFi, BT, Single Core, 240MHz, VRef calibration in efuse, Coding Scheme 3/4
Crystal is 40MHz
MAC: 5c:e5:0c:2b:**:**
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x000bcfff...
Compressed 769888 bytes to 503621...
Wrote 769888 bytes (503621 compressed) at 0x00001000 in 47.9 seconds (effective 128.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
I have a backup - and after its return - everything works.

mmakaay, 2022-05-17:
If you used the `firmware.bin`, then that is not the way in which you can flash the firmware to the lamp using esptool.
Check out [my flashing guide](https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/doc/flashing.md#flash-new-esphome-firmware)
I recommend the use of ESPHome-Flasher, but if you want to flash the firmware using esptool, then follow the instructions in the guide, to make sure that you also flash a bootloader, partition info and such.

Maybe it does work if you flash the plain `firmware-factory.bin` instead, using `write_flash 0x0`, because that contains all required parts too. I never tried this method though, since that factory firmware build output is rather new.

closed 2022-05-17 by ADMiNZ

mmakaay, 2022-05-18:
Happy we got that stitched then 🪡 
Enjoy your lamp!