mmakaay/esphome-xiaomi_bslamp2 #83: Did I brick my bslamp2?

Sakul85, 2022-05-30:
Hello together and thank you all very much for this project and your effort!

After LAN-control was disabled by Xiaomi/Yeelight and broke my home automation, I tried this mod to bring it back to live.

Everything worked well (opening the lamp, soldering wires, making backup of ESP and flash firmware file compiled by esp home dashboard) but finally after everything was done the lamp could not be switched on anymore :-(

Then I restored the original .bin image but it's still not working. I am a bit confused, because the connection and flashing ESP is still working without an error message.

Can anybody help me with this? Should I provide you anthing that can help?

Thank you very much!

[image]

mmakaay, 2022-06-13:
Did you disconnect the GPIO0 to GND connection after flashing?
It's not fully clear in the picture where that lead is going now.
If you have it connected to GND permanently, then the device would only boot into flashing mode, which would result in the symptoms that you described here.

Comforting words coming up:

In general, there is nog much risk in bricking the lamp by flashing. It is always possible to flash a bootloader and application partitions, regardless if earlier flashing attempts failed or not. I have done many broken flash operations on my lamp during development and always was able to restore by means of a direct serial connection for flashing (instead of OTA WiFi).

Sakul85, 2022-06-13:
First of all thank you for trying to help me.

What I get out of the console after removing GPIO from GND and booting is:

```
[21:25:00]rst:0x1 (POWERON_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
[21:25:00]waiting for download
```

mmakaay, 2022-06-13:
Ah yes, joining those ought to be avoided :-) 
I will add it to the docs, thanks!

Is it working now? Then I will close the issue after updating the docs.

Sakul85, 2022-06-13:
Yes, its working. Please close it.

closed 2022-06-13 by Sakul85

mmakaay, 2022-06-13:
And docs have a warning now