mmakaay/esphome-xiaomi_bslamp2 #84: [BUG] Lamp becomes unavailable every hour

szymucha94, 2022-06-19:
**Describe the bug**
Lamp becomes unavailable at least once every hour when on. State "off" is maintained in HA even for days. Lamp is connected to wifi and it does not disconnect.

**To Reproduce**
Turn on the light
Leave it on for 1 hour
Check lamp entity history in HA

**Expected behavior**
Lamp should not be "unavailable" when on and with active wifi connection.

**Additional context**
version used: Firmware: 2022.5.1 (Jun 14 2022, 22:21:11)

[image]

szymucha94, 2022-06-19:
Testing now with updated FW to the one based on 2022.6.1

szymucha94, 2022-06-19:
Same problem on 2022.6.1.

mmakaay, 2022-06-20:
I haven't seen this behavior on any of my three lamps, and with my development lamp,
I did some really long tests with exactly that effect without issues.
Maybe something changed in ESPHome core that now triggers this, so I will flash my dev lamp with the latest ESPHome code and see if I can reproduce this behavior.

One known thing that can trigger this, is excessive logging. The more you log, the less time the WiFi component has to keep up with network traffic. So one thing you could check, is what your logger configuration looks like. From the core config package:

```yaml
# The log level can be raised when needed for debugging the firmware. For
# production, a low log level is recommended. Mainly because high volume log
# output might interfere with the API/WiFi connection stability. So when
# raising the log level, beware that you might see dropped connections from
# Home Assistant and the network log viewer.
logger:
  level: WARN
```

szymucha94, 2022-06-20:
good point, by default I had "info" set. Changed to "warning" and testing this now. Thanks!

mmakaay, 2022-06-20:
Ah, that might already fix the issue for you :-)
Please let me know if this fixes the issue. If not, I'll investige on my lamp.

szymucha94, 2022-06-20:
Unfortunately it still becomes unavailable. 1 time during last hour so far

szymucha94, 2022-06-20:
Okay, I've got something...
lamp disconnected from wifi network. It's interesting because all other active wifi devices (for example two esp powered xiaomi air purifiers and one humidifier with much worse RSSI) remain connected. Also lamp didn't do that on original firmware.
[image]

szymucha94, 2022-06-20:
[image]
Issue does not seem to occur (so far) when special effects are disabled and lamp is just shining blue. I'll let you know after 2-3 hours if it remains stable.

mmakaay, 2022-06-20:
I'm on the train right now, so I can't check for myself (at least not without spooking the cleaning lady who's in my home right now with flashing lights ;-) ), but could you verify if the logging is actually quiet right now? (using the ESPHome dashboard log connection for example)

I know that some extra log messages were added to the light integration, and possibly those are sent out using the `WARN` log level. In that case, tuning the log level to `ERROR` (or leaving the logger out completely if you don't use it) would help.

When the logging is quiet, then tuning down the log level any further wouldn't help.

szymucha94, 2022-06-20:
nope, changing log level to error didn't help. Lamp is still disconnecting every hour in this mode.

mmakaay, 2022-06-21:
I will try it on my lamp. The next to days, I'm a lot away from home, but this weekend I should be able to do a test run.

mmakaay, 2022-06-22:
I had a chance to do some testing, and my lamp is running 100% stable with `DEBUG` level logging enabled.
This means that I get these log messages, every time the fast random effect changes color:
```
[20:39:27][I][light:392]: 'Bedside Lamp Bedroom RGBWW Light' - Keeping current color mode RGB for call without color mode.
[20:39:27][D][light:035]: 'Bedside Lamp Bedroom RGBWW Light' Setting:
[20:39:27][D][light:058]:   Red: 81%, Green: 37%, Blue: 100%
[20:39:27][D][light:084]:   Transition length: 3.0s
```
And even with this, I see no connection drops.

[image]

So I'm afraid I'm unable to reproduce the behavior that you're seeing.

szymucha94, 2022-06-22:
Could you please tell me what is your config?

mmakaay, 2022-06-22:
Basically the example config, and then some extras added (I've got an extra microwave presence sensor built-in).

But I just saw one little hickup after continuously running the fast random effect for a couple of hours. It came back right after that, so I don't think I would have even noticed it had I not added the status sensor.

I have quite a bit of logging, which might be of impact, But when dropouts like this also happen with logging disabled, it's still likely that the device being busy doing stuff other than managing the HA connection / WiFi. This lamp uses an ESP32 that has only one core, which makes it more susceptible to dropping the connection when the device is busy. And with the fast random effect, the CPU is basically busy all the time.
Not sure if I can improve on that.

szymucha94, 2022-06-22:
Thanks for checking. I guess I will work around this, repeat sending the command if lamp disconnects or something.
But if you manage to somehow improve this in the future it'd be nice tho :)

Cheers!

mmakaay, 2022-06-23:
Yeah, that's why I didn't close the issue yet.
I do want to go over the code to see if there are bits that might be hogging the CPU for too long. It's not optimal this way, and not giving up without a fight 😄

szymucha94, 2022-06-30:
One more observation.
Every time lamp disconnects it also turns off the light for about 0,5s or less. At first I thought the algorithm behind picking random color just picks unsupported or problematic setting.

mmakaay, 2022-07-01:
That last thing shouldn't happen. All colors are valid.
So it's definitely related to the reconnecting.

andyboeh, 2022-09-26:
You're probably facing an annoying and long-standig bug in ESPHome that happens on nearly all my ESPHome devices (it never happened on Tasmota, that's why I suspect it's ESPHome). What improved the situation dramatically was adding a separate WiFi AP with WMM disabled. Also, updating my Router's firmware to a newer build (I'm using OpenWrt) improved the situation. 

See https://community.home-assistant.io/t/device-going-unavailable/347769

szymucha94, 2022-09-26:
> You're probably facing an annoying and long-standig bug in ESPHome that happens on nearly all my ESPHome devices (it never happened on Tasmota, that's why I suspect it's ESPHome). What improved the [...]

Hi,
I'm already using openwrt, but with WMM enabled. I will consider creating separate AP directly on the HA hypervisor to mitigate this problem, especially since under newest ESPhome build it disconnects every 30 minutes, not 1h. Thank you!

andyboeh, 2022-09-26:
Yeah, it might also depend on the WiFi chipset: I never experienced this with mt76 radios, only with QCA chipsets.

szymucha94, 2022-10-01:
Nope, with intel chipset, WMM disabled, channel 6 selected and "fast_connect = true" lamp is still reconnecting every <1h.

StybiPL, 2025-01-13:
I am experiencing the same issue. When using the "fast random" effect, the lamp becomes unavailable in Home Assistant after approximately 20–40 minutes. Shortly after, it switches to the "off" state, and the lamp physically turns off.

On all other modes, the lamp works perfectly without any issues.

Let me know if I can assist with additional logs or testing to help resolve this!

szymucha94, 2025-01-13:
@StybiPL
This is esphome bug. Owner of this project won't be able to fix this.
Most likely this wouldn't be happening if we had full esp32-wroom-32d dual core MCUs inside this lamp. Too bad xiaomi uses cheaper single core one.

szymucha94, 2025-02-15:
> [@StybiPL](https://github.com/StybiPL) This is esphome bug. Owner of this project won't be able to fix this. Most likely this wouldn't be happening if we had full esp32-wroom-32d dual core MCUs inside this lamp. Too bad xiaomi uses cheaper single core one.

And just for fun I replaced stock, single core ESP32 with standard dual core module. Flashed with this repo's yaml with one change being the removal of unicore setting. Just started the random color mode - gonna report if it crashes.

[image]

szymucha94, 2025-02-15:
Still crashes :) this is not a problem with resources, ie. interrupting handling of network stack by this program. It's most likely esphome bug.

szymucha94, 2025-02-16:
@mmakaay aaand the decoded stacktrace.
Also fun fact: tried to add ble component just with enable_on_boot parameter (set to false). Esp failed to boot again due to memory depletion, even before wifi component startup.

```
    0x40082b82: panic_abort at /data/cache/platformio/packages/framework-espidf/components/esp_system/panic.c:408
    0x400897b5: esp_system_abort at /data/cache/platformio/packages/framework-espidf/components/esp_system/esp_system.c:137
    0x4008faaa: abort at /data/cache/platformio/packages/framework-espidf/components/newlib/abort.c:46
    0x401443bb: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
    0x40144402: std::terminate() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
    0x4014511d: __cxa_allocate_exception at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_alloc.cc:300
    0x40144434: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54
    0x400dab1f: __gnu_cxx::new_allocator >::allocate(unsigned int, void const*) at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/ext/new_allocator.h:111
    0x400dab1f: std::allocator_traits > >::allocate(std::allocator >&, unsigned int) at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/alloc_traits.h:436
    0x400dab1f: std::_Rb_tree, std::less, std::allocator >::_M_get_node() at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_tree.h:588
    0x400dab1f: std::_Rb_tree_node* std::_Rb_tree, std::less, std::allocator >::_M_create_node(esphome::light::ColorMode const&) at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_tree.h:642
    0x400dab1f: std::_Rb_tree_node* std::_Rb_tree, std::less, std::allocator >::_Alloc_node::operator()(esphome::light::ColorMode const&) const at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_tree.h:556
    0x400dab1f: std::_Rb_tree_iterator std::_Rb_tree, std::less, std::allocator >::_M_insert_, std::less, std::allocator >::_Alloc_node>(std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, esphome::light::ColorMode const&, std::_Rb_tree, std::less, std::allocator >::_Alloc_node&) at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_tree.h:1776
    0x400dab1f: std::_Rb_tree_iterator std::_Rb_tree, std::less, std::allocator >::_M_insert_unique_, std::less, std::allocator >::_Alloc_node>(std::_Rb_tree_const_iterator, esphome::light::ColorMode const&, std::_Rb_tree, std::less, std::allocator >::_Alloc_node&) at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_tree.h:2229
    0x400dab1f: void std::_Rb_tree, std::less, std::allocator >::_M_insert_unique(esphome::light::ColorMode const*, esphome::light::ColorMode const*) at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_tree.h:2475
    0x400dad18: std::set, std::allocator >::set(std::initializer_list, std::less const&, std::allocator const&) at /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_set.h:247
    0x400dad18: esphome::light::LightCall::get_suitable_color_modes_() at /data/build/lr-bedside-lamp/src/esphome/components/light/light_call.cpp:446
    0x400dae3e: esphome::light::LightCall::compute_color_mode_() at /data/build/lr-bedside-lamp/src/esphome/components/light/light_call.cpp:387
    0x400daf06: esphome::light::LightCall::validate_() at /data/build/lr-bedside-lamp/src/esphome/components/light/light_call.cpp:146
    0x400db803: esphome::light::LightCall::perform() at /data/build/lr-bedside-lamp/src/esphome/components/light/light_call.cpp:33
    0x400e4685: esphome::light::RandomLightEffect::apply() at /data/build/lr-bedside-lamp/src/esphome/components/light/base_light_effects.h:98
    0x400dbd41: esphome::light::LightState::loop() at /data/build/lr-bedside-lamp/src/esphome/components/light/light_state.cpp:117
    0x40155ddd: esphome::Component::call_loop() at /data/build/lr-bedside-lamp/src/esphome/core/component.cpp:77
    0x40155e79: esphome::Component::call() at /data/build/lr-bedside-lamp/src/esphome/core/component.cpp:104
    0x400e03b8: esphome::Application::loop() at /data/build/lr-bedside-lamp/src/esphome/core/application.cpp:74
    0x400e3d2e: loop() at /data/packages/03087324/packages/activate_preset_svc.yaml:37
    0x400d88ea: esphome::loop_task(void*) at /data/build/lr-bedside-lamp/src/esphome/components/esp32/core.cpp:69
```

szymucha94, 2025-02-16:
Just to confirm: every action (changing brightness, switching colors, changing color temperature) results in more and more leaking memory. Until the crash.

[image]

closed 2025-02-16 by szymucha94