mmakaay/esphome-xiaomi_bslamp2 #85: mDNS / HomeKit

8times4, 2022-06-21:
Hey mmakaay,

I love the work that you have done on this project. 
Could you please tell me if the mDNS service is kept around for HomeKit to pick up? 

Thank you,
8x4

khmm12, 2022-06-21:
Short: there is mDNS service, but HomeKit won't pick up a device.

Long: ESPHome has mDNS component out-of-box to device be discovered by ESPHome toolchain and HA integration. But ESPHome doesn't have a component to communicate with HomeKit and HomeKit doesn't have an integration to communicate with ESPHome.

closed 2022-06-21 by 8times4

khmm12, 2022-06-21:
@8times4 you're welcome! 

You can use [HA HomeKit](https://www.home-assistant.io/integrations/homekit/) component which works great.

8times4, 2022-06-22:
Yeah, thanks @khmm12 , I was planning to use HA, but would be nice to not rely on that for HomeKit to work. 
I was eyeing with https://github.com/HomeSpan/HomeSpan as this could be an option to add "native" HomeKit support.