mmakaay/esphome-xiaomi_bslamp2 #88: [FEATURE] Chinese Translation

Thisisnotgoingpublished, 2022-06-30:
**Is your feature request related to a problem? Please describe.**
I want to translate this project. I find it's annoying me.

**Describe the solution you'd like**
I offering the translation,

**Describe alternatives you've considered**
Google Translate

Thisisnotgoingpublished, 2022-06-30:
README.zh.md

# 对小米®米家床头灯 2 进行 ESPHome 兼容支持

[ [变更记录](CHANGELOG.md) | [许可协议](LICENSE.md) | [行为准则](CODE_OF_CONDUCT.md) ]

小米®米家床头灯2是一款智能型三色二温(EGBWW) 的 LED 照明灯具，由易来(Yeelight)公司替小米®米家品牌代工。这款灯具可以通过Wi-Fi网络以及该机器前侧的触摸面板来对其进行操控。触摸面板带有电源键、变色键还有一个控制灯光亮度的滑条。

本项目为用户提供客制化后的 ESPHome 组件，使得用户对本照明灯具进行全方位的完全控制，并把本照明灯具加入到您的家居助手(Home Assistant)里成为可能。

## 功能

* 本照明灯具 ** 与家居助手(Home Assistant)集成十分容易** 只需要采用ESPHome加入就可以做到。

* **本照明灯具不再会与米家云(Mijia Cloud)进行通信。**采用本固件后，用户可以完全放心，网络传输仅限于使用者的本地网络。这一点也与家居助手(Home Assistant)的理念相吻合，那就是为使用者提供本地家居自动化平台，以个人隐私为重中之重。

* **不用本地局域网控制选项** 就能把照明灯具加入到家居助手(Home Assistant)里。这点非常重要，是因为小米不知道是犯了什么糊涂穷尽了全体成员的智慧拍脑门子决定要移除本机器的本地局域网控制功能，这样极大地破坏了现有的集成环境。

* **夜灯功能支持多重颜色**。原固件只支持单一的暖白夜灯颜色。

* **平滑的灯光颜色切换**，不像是当前易来(Yeelight)集成版本的那样。Homekit集成的确提供了挺不错的切换，可是在我个人的系统中，在家居助手(Home Assistant)用户界面里缺少色温白光模式。

* **由于本照明灯具的组件是被当做ESPHome组件存在的，用户不需要依照该灯具过去的运行功能来使用。**你可以把本照明灯具依照你的偏好添加到你的家居自动化环境里。使用滑条控制您音响系统的音量你喜欢么？长按电源按键把整个屋子变成夜晚模式可以吗？使用滑条的量度显示面团发酵进度怎么样？好哇，说走就走！ (^U^)

* **所有照亮前面板的LED(电源键、变色键和滑条的10颗LED)可以一对一控制。**这意味着你有12颗可以依照你的偏好使用的LED，而不需要依照该灯具过去的运行功能来使用。
* **通过硬件模改以扩展机器功能可能性** 存在[GPIO 针没有被使用](doc/technical_details.md#esp32-pinout)。如果你的人送外号是“修改狂人”，你可以利用这些针来想出你自己的硬件妙用来扩展机器的功能。

## 快速开始指导

对刷ESPHome到机器中有经验的人可以来看看：

* 确保你使用的是ESPHome 2021.10.0或者更加新的版本。
* 复制[`example.yaml`](example.yaml)到`<CONFIG_DIR>/<NODE_NAME>.yaml`。
* 根据需求自行修改配置文件(详阅 [configuration guide](doc/configuration.md))。
* 编译`firmware.bin`文件并下载到被你连接到串行转USB适配器的机器里。
* [拆灯](doc/flashing.md#opening-the-lamp-to-expose-the-pcb) 连接灯具的`TX`, `RX`, `GND` 和 `GPIO0` 调试接点到串行适配器。请看[接点的位置图](doc/images/09_debug_pads_for_soldering.jpg)。
* 给本照明灯具通电，`GPIO0`要与GND连在一起以启用刷机模式。
* 把`firmware.bin`刷进机器，例如使用[esphome-flasher](https://github.com/esphome/esphome-flasher)。
  
##升级

我总想让这个机器不用其他修改就可以升级固件。但有些时候，根本不可能。发生了的话，升级说明在这里

* [升级到 2021.10.0](doc/upgrading_to_2021.10.0.md)

## 内容一览表

* [何苦要用自制固件？](doc/why_custom_firmware.md)
* [安装指导](doc/installation.md)
* [配置指导](doc/configuration.md)
* [刷机指导](doc/flashing.md)
* [技术详情](doc/technical_details.md)
* [资助](doc/sponsoring.md)

mmakaay, 2022-06-30:
Xiaomi and Yeelight annoying you is not good :-)
I'm not sure what the best way would be to get the translated documentation in the project.
Maybe a `doc_cn` folder would be an option?

Are you translating all of the project, or mainly the starting page?
The configuration guide might be the one least useful for translation, since that one will get some more changes over time. Both additionals and breaking changes. The rest of the pages might be quite static at this point.

Thisisnotgoingpublished, 2022-07-01:
> Xiaomi and Yeelight annoying you is not good :-) I'm not sure what the best way would be to get the translated documentation in the project. Maybe a `doc_cn` folder would be an option?
>
> Are you [...]

I'll translate the whole project, but it will take time, which I am not have quiet amount of that precious. And I have to do some updates from time to time here and there. I promise that will not use machine to translate it. But I am also unexperienced. So please bear with me till I get it all done.

For the time being, I am just gonna leave all my work right here.

The Format Being Like:

File path
========
Content

Thisisnotgoingpublished, 2022-07-01:
doc\flashing.zh.md

< [配置指导](configuration.md) | [主页](../README.md) | [技术详情](technical_details.md) >

# 刷机指导

目录:
* [警告](#warning)
* [所需材料](#tools-needed)
* [拆灯，露出PCB板](#opening-the-lamp-to-expose-the-pcb)
* [把转接线焊到主板](#solder-wires-to-the-board)
* [把转接线连接到用户的串行转USB适配器上](#connect-the-wires-to-your-serial-to-usb-adapter)
* [用户的USB适配器只有一个GND针怎么办](#when-you-only-have-one-gnd-pin-on-your-usb-adapter)
* [备份当前固件](#make-a-backup-of-the-current-firmware)
* [如何恢复已备份的固件](#how-to-restore-the-backed-up-firmware)
* [刷入新的 ESPHome 固件](#flash-new-esphome-firmware)
* [刷机故障排除](#troubleshooting-flash)

## 警告

这里的指导都是我们用心撰写的，不过我们不能给用户任何保证。也许在操作途中用户会弄坏用户的照明灯具和用户的电脑。

## 所需工具

* 六角扳手 (2mm, 5/64") 或是梅花形 (T8) 螺丝刀
* 焊接烙铁
* 串行转USB适配器 (例如 FTDI) 可以提供 3.3V 的收发(RX/TX)信号
* 一些转接线
* 可选工具: 透明胶带、热胶枪、放大镜

## 拆灯，露出PCB板

*提示: 读者可以任意点击下方图片以查看完整尺寸图像.*

移除照明灯具底部的胶垫, 可以看到照明灯具底部的连接灯具主体的4个螺丝孔。

[image][image]

并不需要把胶垫全部剥离。只需要足够看到螺丝孔，其实那就够了。

[image]

把胶垫隐藏起来的4个螺丝卸下来。

[image][image]

把照明灯具底部和灯具其他部分拆开，露出PCB板。可能得稍微用点力才行。一点一点的把底座拉下来，一直到底部脱落为止。

[image][image]

[image]

## 把接线焊到主板

应该把接线连接到调试接触点，如下图所示。

[image]

许多串行转USB适配器有一些抬头针，让用户可以把接线接到设备上（并不需要焊接操作）。于是乎，使用杜邦线(dupont wire)可能更加实用。把转接线尾部掐折、拨线、上焊锡、把接线接到主板上。

*注意: 到底是使用插入式还是被插式线头这点取决于你想怎么接在串行适配器上。在本范例中，我使用的是插入式转接线，这样我可以把线直接插入适配器板上。*

[image]

焊接转接线到 `RX`, `TX`, `GND` 和 `GPIO0` 调试接点.

*注意: 这个板上游一个调试接点提供3.3V电。不要采用本接点给适配器供电。一定要确保照明灯具的电是从照明灯具自己的电源供应供电。*

一些提示:

[... 162 lines cut ...]
[image]

From here on, it is possible to flash the lamp OTA (over the air, which means that the firmware is
uploaded over WiFi) from ESPHome. Therefore, it is now time to tuck away or remove those soldered
wires.

Because I want to keep them around for future use, I tuck them away, making sure that the connectors
don't touch each other or the board.

[image]

The bottom cover can now be put back on. The lamp is ready for use.

[image]

## Troubleshooting flash

If you have **A fatal error occurred: MD5 of file does not match data in flash!**, then make sure
you are powering the board using the lamp's own power adapter. We've seen these errors when trying
to power the board using the 3.3V debug pad.

After seeing this error, user @tabacha was able to successfully flash his lamp using the regular
power adapter.

< [Configuration guide](configuration.md) | [Index](../README.md) | [Technical details](technical_details.md) >

mmakaay, 2022-07-01:
I thinks it's easier to get it done through pull requests. Copying and pasting from here might end up in errors.
Pull requests might sound difficult, but GitHub has a great feature, especially for editing documentation. You can go to a documentation page and there you can click on:

[image]

"edit this file".

After editing, you can do "Propose changes" at the bottom, to send in a pull request, which I can then accept.

I added some files to the project tree, so you can work with those:

- [README.chinese.md](https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/README.chinese.md)
-[ doc.chinese/*](https://github.com/mmakaay/esphome-xiaomi_bslamp2/tree/dev/doc.chinese)
 
I updated the internal links, so the README.chinese.md links point to the files inside the doc.chinese/ folder and the images are used from the English doc/images/ folder.