mmakaay/esphome-xiaomi_bslamp2 #91: [QUESTION]

martinrevert, 2022-09-20:
First of all, I think this is not a bug, maybe it is just me not doing things right.

I compiled ESPHome using old command line tools which worked fine, and produced ok firmwares for my 2 bslamp2. 

I uploaded the frimware to both lamps, and they were discovered OK by Home Assistant.

The problem was integrating them and trying to update them via EPSHome Dashboard on Home Asssitant. I literally bricked them trying an update and had to reupload the ESPHome firmware-factory.bin previously generated to recover (thank God!) them.

So, the question is: How do I integrate this firmware into ESPHome Dashboard of Home Assistant? Could be the problem I faced that wasn't including the appropiate toolchain on the config yaml? I didn't need to inform that to compile locally, but maybe is needed to recompile using EspHome Dashboard?

The expected beahvior for me is after I upload the firmaware, to integrate into ESPHome Dashboard and from that, administrate updates from there.

Sorry for newbiness!

mmakaay, 2022-10-08:
First: bricking isn't really a risk. You can always get the device to upload new working firmware from your command line install, using the serial connection. So no worries about that.

If you're running into the aarch64 problem, how are you using the ESPHome Dashboard of Home Assistant? I don't really understand how you would get a (broken) firmware to upload in that case, since the build would fail.

martinrevert, 2022-11-19:
Hi @mmakaay  , I wasnt't aware about your kind response, thanks fro taking the time. Yes, I was trying to use ESPHome from Home Assistant dashboard over aarch64, I didn't knew that there's a problem with it,, but probably also I made a mistake providing the correct toolchain in that yaml, so the results where maybe incosistent for you, because my probable mistake there. For now I can live with the firmware I produced from ESPHome 2022.8.3 command line, it works, the only tiny problem is that the lamps are not present in ESPHome dashboard, but nothing I can't live without. You know, the missus wants lamps working, she don't care about the latest firmware hahaha. I close the issue because it has nothing to do with you, but i'm interested in following the aarch64 problem..where can I add my +1? Is there an issue on ESPHome github to follow? Thanks!

closed 2022-11-19 by martinrevert

mmakaay, 2022-11-19:
There is an issue for it at:
https://github.com/esphome/issues/issues/3076

I recently looked into the issue, and added some comments to that issue describing my findings. I did get pretty close to the culprit, but got stuck at finding a solution.