mmakaay/esphome-xiaomi_bslamp2 #92: Flashing issue - Failed to connect[BUG]

ideczynieide, 2022-09-26:
When trying to flash I'm getting the following message:

_Using 'COM4' as serial port.
Connecting......................................
Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting_

This makes me think that clearly there's no communication, however when I do "View logs" it appears connection is made successfully:
_Using 'COM4' as serial port.
Showing logs:
[08:04:22]07:04:21.000 [I] miio_monitor: work time = 200s, memory left = 64960
[08:04:27]07:04:26.090 [I] ots: -->kplv sent.
[08:04:28]07:04:26.480 [I] ots: <--kplv ack.
[08:04:32]07:04:31.000 [I] miio_monitor: work time = 210s, memory left = 64960
[08:04:42]07:04:41.000 [I] miio_monitor: work time = 220s, memory left = 64960
[08:04:44]07:04:42.480 [I] ots: -->kplv sent.
[08:04:44]07:04:42.860 [I] ots: <--kplv ack.
[08:04:47]07:04:46.000 [I] miio_monitor: work time = 225s, memory left = 64940
[08:04:48]07:04:47.000 [I] miio_monitor: work time = 226s, memory left = 64960_

**To Reproduce**
1. Plug in Lamp's power and wait 20s
2. Plug USB programmer to PC (I am using this https://www.amazon.pl/dp/B08T24NML9?ref_=pe_25095781_587346921_E_301_dt_1&th=1)
3. View logs in ESPHome-Flasher-1.4.0-Windows-x86
4.Close ESPHome flasher because it seems I cannot immediately flash due to COM port being locked by previous "view logs" action
5. Open ESPHome flasher, pick .bin file and click flash
6. Programmer's red TX led flashes
7. I get reported error msg.

I also tried ESP flasher x64, different combination of reboots, USB ports, waiting more for lamp to boot etc. If USB programmer is plugged early then I am able to see boot log
[bslamp2 log.txt](https://github.com/mmakaay/esphome-xiaomi_bslamp2/files/9643460/bslamp2.log.txt)

andyboeh, 2022-09-26:
Did you connect GPIO0 to GND? Seems like your device never enters flashing mode.

ideczynieide, 2022-09-26:
I did - but then I didn't triple check this option. Thanks, I'll give this a try.

mmakaay, 2022-10-08:
Did the hint by @andyboeh help?
His theory was correct I would say. If GPIO0 is properly grounded when booting the device, then you should not see the startup logs from your bslamp2 log.txt file.

ideczynieide, 2022-11-17:
Hi - it did indeed. Thank you for helping out!

closed 2022-11-17 by ideczynieide