mmakaay/esphome-xiaomi_bslamp2 #96: [FEATURE] Add BTProxy

TheFes, 2022-10-25:
**Is your feature request related to a problem? Please describe.**
As HA now supports Bluetooth proxies, and the Xiaomi Bedside lamp 2 comes with an ESP, I thought it would be nice if it would also act as a Bluetooth proxy.
I tried adding it myself, which results in a successful OTA flash, but then the light doesn't work anymore. I had to flash a non-btproxy version using the ftdi-adapter again to make it work. So this could be some kind of hardware limitation, it could also be that I'm doing something wrong :)

This is what I added:
```yaml
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
```

**Describe the solution you'd like**
The lamp to act as a BT Proxy

**Describe alternatives you've considered**
Adding it myself, without success..

mmakaay, 2022-11-03:
It would be a really nice feature, but I highly doubt that the hardware will support this, being a single-core ESP32 and not a standard one. Maybe only enabling the tracker with wifi disabled would work, but that is not a practical solution.

andyboeh, 2022-11-18:
I tried to enable BLE as well and it wouldn't work. If I recall correctly, the BLE stuff took to long and the watchdog resets the device. You can capture a log if you attach serial console.
I did not bother trying to disable the watchdog or reconfiguring it for a longer timeout and just used another ESP for it. I'm also quite certain that the reason is the single-core design of this particular ESP32.

TheFes, 2022-11-18:
Ah clear! I'll close the issue then as it is a hardware limitation, so it's not possible :)

closed 2022-11-18 by TheFes