mmakaay/esphome-xiaomi_bslamp2 #99: [QUESTION]

ideczynieide, 2022-11-20:
I flashed one lamp which works great. However... I got another lamp because this integration works really well, flashed this one too and here's the issue:
- Flashing successful
- My Home Assistant server is on 192.168.0.X   network, but lamps are on 192.168.20.X
- First lamp I flashed works fine and is seen by HA
- Second is not visible from 0.X network (e.g. ping - request time out, HA doesn't see it). It is however visible from .20.X network
- Both have same settings with just name, friendly_name and IP being different

I reflashed a bunch of times thinking this is it but that isn't helping. I would think this is my router (Unifi) config but then configurations of lamps are the same. Any thoughts?

mmakaay, 2022-11-21:
The question is not specifically related to the lamp firmware, but let's see if we can find a solution.

When you can ping the second lamp from the 20.X network, but not from the 0.X network, then it sounds like a networking issue.

- Do you give it a static IP address or does it retrieve one through DHCP?
- If you use a static IP, did you provide it with the correct default gateway? Without a default gateway assigned, the lamp wouldn't know how to return ping responses when the source of the ping is coming from a different subnet.
- Are you trying to add the lamp to HA by entering its hostname/IP-address when adding the integration?
- If not, are you expecting the lamp to be automatically discovered? That would not work, unless the Unifi router performs mDNS proxying. 

These are the first things that come to mind, looking at your symptoms.

ideczynieide, 2022-11-23:
Hi, thanks for your help!

Here's lamp's config. I am assigning static IP. I tried with gateway set to 192.168.20.1 but that didn't change anything. And besides working lamp has this set as 192.168.0.1.
wifi:
  ssid: ssid
  password: pass
  manual_ip:
    static_ip: 192.168.20.253
    gateway: 192.168.0.1
    subnet: 255.255.255.0

randybb, 2022-11-23:
Are you expecting that the packets will then magically jump to the other network as the gateway is not within range of your network 192.168.20.0/24?
I don't expect that you are going to bring mDNS to work (it can be done, but you have to know what you are doing). Just create a static record on your DHCP server, so this device will get always the same address. Since mDNS in not going to work between networks you have add it to HA manually (no auto-discovery) using its IP address.
If you are doing any filtering between these networks, make sure you are allowing tcp/6053 from HA to 192.168.20.0/24 as esphome device is a "server" and HA is the "client" which initiates connection.

mmakaay, 2022-11-23:
Like @randybb says, the problem lies with the invalid gateway IP address.
Given the 192.168.20.0/24 subnet, the gateway ought to be in the 192.168.20.1 - 192.168.20.254 range. The device will not be able to reach the router at 192.168.0.1 and packets therefore cannot be sent from the lamp to Home Assistant.

You have another lamp that does work. I bet it has a gateway IP address like 192.168.20.1 instead.

Once you have fixed the gateway IP address, you should be able to add the integration to Home Assistant manually, using static IP address 192.168.20.253.

I have played with ESPHome devices in a separate VLAN + subnet, but because of the extra moving parts (and my trust in ESPHome devices), I switched to hosting the devices in my regular user VLAN. Most devices have a fixed IP, because OCD, but I also have a few running based on dynamic IP without issues. That is the goodness that mDNS will get you.

randybb, 2022-11-23:
A fixed address on devices is not a flexible solution. You can still have them nicely sorted, just use DHCP reservation and your OCD will be happy. Because if you want to move a device to other network, or do any changes, you have to recompile, which is crazy :) In the ideal world once you configure your lamp, you will never need to touch it. This is why I prefer DHCP in customer networks (I am a network engineer) and at home :) And with unifi solution only the controller and gateway (which is DHCP server) need to have static addresses.

ideczynieide, 2022-11-24:
Thanks as always everyone. I removed any fixed IP config, reserved with DHCP and it works!
For some reason ESPHome addin to Home Assistant doesn't see this device as 'online' (any thoughts why?) unlike it did with the other lamp but I was able to add this in manually.

randybb, 2022-11-24:
ESPHome dashboard uses mDNS, which is again not available in your case :) You need to enable the ping option in the Home Assistant add-on and  set `use_address:` https://esphome.io/components/wifi.html to tell dashboard what is it's address.

closed 2022-11-24 by ideczynieide