mmakaay/esphome-xiaomi_bslamp2 #101: some help needed

kerniger, 2022-11-23:
hi,
thanks for your work.
I am stuck in ESPHome. I installed it as an Add-on on HASS on rpi4.
I edited the example.yaml file and put it in \config\esphome\
The xiaomi_bslamp2 folder i put in \config\esphome\custom_components.
Now, when i try to validate i get a lot of errors:

> INFO Reading configuration /config/esphome/mann.yaml...
ERROR Unable to import component xiaomi_bslamp2.light:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/custom_components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.text_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/custom_components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.output:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/custom_components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/custom_components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/custom_components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/custom_components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
WARNING 'nachttisch_mann': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
Failed config

Can you help me?

mmakaay, 2022-11-23:
You do not have to download anything for this to work. The config makes use of the newer external components support, which lets ESPHome download the required files from this GitHub repo.

Therefore, editing the example.yaml ought to be enough. Please delete the custom_components download from your host and try again.

The related instructions can be found in my documentation section:
https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/release/2021.10.0/doc/installation.md

mmakaay, 2022-12-23:
No further updates received, closing the issue.
Feel free to reopen when more support is needed.

closed 2022-12-23 by mmakaay