mmakaay/esphome-xiaomi_bslamp2 #102: Unexpected error: could not open port 'COM1': PermissionError(13, 'Access is denied.', None, 5)

RcKVQhBWO, 2022-11-27:
In esphome-flasher for windows,
when I click show log and flash esp I get an error:
"Using 'COM1' as serial port.
Unexpected error: could not open port 'COM1': PermissionError(13, 'Access is denied.', None, 5)"

Same error on windows 10 and Debian 
sometimes when I turn power to lamp on log will show:
rst:0x1 (POWERON_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download

however I cannot flash, if I press show log it will revert to permission Error.

  I have various times tried:
-  setting bits per second in device manager to 9600 and 115200 
- opening esphome flasher as administrator 
-  2 different usb data cables 
-  installed drivers on windows for ftdi232RL 
- restarting pc
- somewhat verified solder points are ok by observing rx and tx led state change on usb converter when powering on lamp ?
- in Arduino IDE serial monitor I can consistently observe  log showing:   "rst:0x1 (POWERON_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download" when I plug power cable to lamp in.
- verified fdti switch is in 3.3v selection 
- I will be able to try different usb converters in afew days 

If possible could you explain the exact order/ chain of events I should do in regards to connecting power and usb cables
And any suggestions? 
Thanks for your time

mmakaay, 2022-12-10:
I do get this error a lot myself when flashing from windows.
This is not a problem with ESPHome or the lamp as such, but it happens before the flasher is even trying to send data.
It looks like another process on the computer might be grabbing the comport and therefore locking it for other use. I never found what program or system component was the culprit.

My solution: When flashing, I connect GPIO0 to ground, then I plug in the lamp power, then I plug in the USB cable into the computer and right afterwards I start the flashing operation. When the permission denied occurs, I unplug and replug the USB and try to start the flashing again.

This procedure hasn't failed me yet. I hope you can get it to work this way too.

RcKVQhBWO, 2022-12-11:
Hi thanks for that valuable information.
Following your solution methodology has helped however now afew seconds after clicking flash ESP I get this error: 
"Unexpected error: ESP Chip Auto-Detection failed: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting"

I have an unsettling feeling that my rx solder is not great so I will redo that and see if that helps

RcKVQhBWO, 2022-12-11:
I was about to go re solder but decided to try one more time and flashing was successful! 
Thanks for your help 
Now to figure out why logs are saying no matching WiFi found :)