mmakaay/esphome-xiaomi_bslamp2 #105: [BUG] Broken TX pad

hellcry37, 2022-12-22:
**Describe the bug**
Broken TX pad, where I can get it back to flash my lamp? 

**To Reproduce**
Steps to reproduce the behavior:
1. Is best not to reproduce my mistake :)

**Expected behavior**
Expect maybe to find another point where I can do TX.

**Additional context**
Add any other context about the problem here.

[image]

mmakaay, 2022-12-23:
You already found GND, but for completeness: the easiest accessible GND connection is likely the big patch:

[image]

For RX, I think you'd have to take out the PCB and turn it over. 
There you'll find the ESP32. The pinout is documented in the technical document of the project.
Here's a screenshot of the part in which TX and RX are:

[image]

I am not sure if RX emerges the PCB in other spots than the ESP32 pins and the debug pads. Worst case scenario, you'd have to connect to the RX pin of the ESP32 itself. But given the soldering skills, that might turn out to be a bit of a challenge ;-) Maybe it's something that you can do by manually pressing a solid core wire against the RX pin, holding it there during the flashing process.

If you do decide to solder on the RX pin, make sure you also have a vacuum pump and some desoldering wick to remove excessive solder in case a blob of solder creates a bridge between RX and neighbouring pins. Also don't hold the soldering iron agains the pin for too long, so you don't overheat the hardware.
For soldering leads this way (I had to do so to be able to check the signals that go to the LED hardware) I normally would use a solid core wire, put a little bit of solder on the wire tip and then us that to connect it to the chip's pin. It's not the recommended way of soldering for permanent connections, but for temporary goals like this, it's the one with the least hassle for me.

I will see if I can find RX emerging somewhere else by probing the board of my dev lamp.

hellcry37, 2022-12-23:
Haha i doubt I can desolder that ESP chip but if I can at least put a pin on it maybe, just maybe. Thanks

closed 2022-12-23 by hellcry37

mmakaay, 2022-12-23:
Yeah, don't desolder the full chip! :-D
The pins are accessible from the outside, so that's the best option if every thing else fails.