mmakaay/esphome-xiaomi_bslamp2 #110: Fail to compile ESPHome 2023.2.1 update

rafjog, 2023-02-16:
With the ESPHome 2023.2.1 update, it fails to compile, showing the following errors:

WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
Failed config

api: [source /config/esphome/.esphome/packages/c1741637/packages/core.yaml:49]
  reboot_timeout: 0s
  services: 
    - service: activate_preset
      variables: 
        my_group: string
        my_preset: string
      then: 
        - 
          Too many candidates found for 'presets_id' type 'xiaomi::bslamp2::PresetsContainer' You must assign an explicit ID to the parent component you want to use.
          preset.activate: 
            group: !lambda |-
              return my_group;
            preset: !lambda |-
              return my_preset;
  port: 6053
  password: ''
light.xiaomi_bslamp2: [source /config/esphome/.esphome/packages/c1741637/packages/behavior_default.yaml:62]
  platform: xiaomi_bslamp2
  id: my_light
  name: Bedside Lamp RGBWW Light
  default_transition_length: 800ms
  on_brightness: 
    - then: 
        - globals.set: 
            id: current_brightness
            value: !lambda |-
              return x;
        - script.execute: 
            id: update_front_panel
  effects: 
    - random: 
        name: Slow Random
        transition_length: 30s
        update_interval: 30s
    - random: 
        name: Fast Random
        transition_length: 3s
        update_interval: 3s
  presets: 
    rgb: 
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      red: 
        red: 1.0
        green: 0.0
        blue: 0.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      green: 
        red: 0.0
        green: 1.0
        blue: 0.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      blue: 
        red: 0.0
        green: 0.0
        blue: 1.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      yellow: 
        red: 1.0
        green: 1.0
        blue: 0.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      purple: 
        red: 1.0
        green: 0.0
        blue: 1.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      randomize: 
        effect: Fast Random
    white: 
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      cold: 
        color_temperature: 153.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      chilly: 
        color_temperature: 275.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      luke: 
        color_temperature: 400.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      warm: 
        color_temperature: 588.0
  disabled_by_default: False
  restore_mode: RESTORE_DEFAULT_OFF
  gamma_correct: 2.8
  flash_transition_length: 0s
light.xiaomi_bslamp2: [source /config/esphome/bslamp2.yaml:58]
  platform: xiaomi_bslamp2
  name: My Bedside Lamp
  id: my_bedside_lamp
  default_transition_length: 500ms
  effects: 
    - random: 
        name: Randomize
        transition_length: 3s
        update_interval: 3s
  on_brightness: 
    - then: 
        - logger.log: 
            format: The brightness changed!
            level: DEBUG
            args: []
            tag: main
  presets: 
    my_color_presets: 
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      red: 
        red: 1.0
        green: 0.0
        blue: 0.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      green: 
        red: 0.0
        green: 1.0
        blue: 0.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      blue: 
        red: 0.0
        green: 0.0
        blue: 1.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      yellow: 
        red: 1.0
        green: 1.0
        blue: 0.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      purple: 
        red: 1.0
        green: 0.0
        blue: 1.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      randomize: 
        effect: Randomize
    my_white_presets: 
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      cold: 
        color_temperature: 153.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      chilly: 
        color_temperature: 275.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      luke: 
        color_temperature: 400.0
      
      Too many candidates found for 'id' type 'xiaomi::bslamp2::XiaomiBslamp2LightState' Some are 'my_light', 'my_bedside_lamp'.
      warm: 
        color_temperature: 588.0
  disabled_by_default: False
  restore_mode: RESTORE_DEFAULT_OFF
  gamma_correct: 2.8
  flash_transition_length: 0s
binary_sensor.xiaomi_bslamp2: [source /config/esphome/.esphome/packages/c1741637/packages/behavior_default.yaml:130]
  platform: xiaomi_bslamp2
  id: my_color_button
  for: COLOR_BUTTON
  on_multi_click: 
    - timing: 
        - state: True
          min_length: 0ms
          max_length: 600ms
      then: 
        - 
          Too many candidates found for 'presets_id' type 'xiaomi::bslamp2::PresetsContainer' You must assign an explicit ID to the parent component you want to use.
          preset.activate: 
            next: preset
      invalid_cooldown: 1s
    - timing: 
        - state: True
          min_length: 600ms
      then: 
        - 
          Too many candidates found for 'presets_id' type 'xiaomi::bslamp2::PresetsContainer' You must assign an explicit ID to the parent component you want to use.
          preset.activate: 
            next: group
      invalid_cooldown: 1s
  disabled_by_default: False
  name: my_color_button
  internal: True
text_sensor.xiaomi_bslamp2: [source /config/esphome/.esphome/packages/c1741637/packages/behavior_default.yaml:166]
  
  Too many candidates found for 'output_id' type 'xiaomi::bslamp2::XiaomiBslamp2LightOutput' You must assign an explicit ID to the parent component you want to use.
  platform: xiaomi_bslamp2
  name: Bedside Lamp Light Mode
  id: my_light_mode
  disabled_by_default: False

mmakaay, 2023-02-17:
It looks like you have a duplicate definition of things like the light component and bslamp2 component.
So my guess here is that you have copied stuff from `core-behavior.yaml` into your device's yaml configuration file, but left the package `packages/behavior_default.yaml` in the `packages:` section. As a result, things are now defined twice, causing the `You must assign an explicit ID to the parent component you want to use` errors.

mmakaay, 2023-02-17:
BTW: I tried compiling on the latest ESPHome and it worked, except for MQTT, which failed with a compile error. That seems to have broken in the ESPHome mqtt implementation (probably due to framework updates). But once I deleted the mqtt component from my config, I could compile.

rafjog, 2023-02-17:
Ok, thanks, I'll take a look, the strange thing is that with previous versions of ESPHome, it compiled and I was able to install on the lamp.

rafjog, 2023-02-17:
My code is:
substitutions:
  name: bedside-lamp
  friendly_name: Bedside Lamp
  light_name: ${friendly_name} RGBWW Light
  light_mode_text_sensor_name: ${friendly_name} Light Mode
  default_transition_length: 800ms

packages:
  bslamp2:
    url: https://github.com/mmakaay/esphome-xiaomi_bslamp2
    ref: dev
    files:
      - packages/core.yaml
      - packages/behavior_default.yaml
      - packages/ota_feedback.yaml
      - packages/activate_preset_svc.yaml
    refresh: 0s

wifi:
  networks:
    - ssid: !secret wifi_ssid          
      password: !secret wifi_password

api:

ota:

light:
  - platform: xiaomi_bslamp2
    name: My Bedside Lamp
    id: my_bedside_lamp
    default_transition_length: 0.5s
    effects:
      - random:
          name: Randomize
          transition_length: 3s
          update_interval: 3s
    on_brightness:
      - then:
          - logger.log: The brightness changed!
    presets:
      my_color_presets:
        red:         { red: 100%, green: 0%,   blue: 0%   }
        green:       { red: 0%,   green: 100%, blue: 0%   }
        blue:        { red: 0%,   green: 0%,   blue: 100% }
        yellow:      { red: 100%, green: 100%, blue: 0%   }
        purple:      { red: 100%, green: 0%,   blue: 100% }
        randomize:   { effect: Randomize                  }
      my_white_presets:
        cold:        { color_temperature: 153 mireds      }
        chilly:      { color_temperature: 275 mireds      }
        luke:        { color_temperature: 400 mireds      }
        warm:        { color_temperature: 588 mireds      }

deleting packages/behavior_default.yaml keeps giving errors

mmakaay, 2023-02-17:
Can you post your configuration surrounded by 

\`\`\`
three backticks
before and after the code
\`\`\`

You can also paste the code, select it and then click the `< >` icon in the editor toolbar.
That will keep the structuring intact and makes it easier to try out your configuration on my system.

rafjog, 2023-02-18:
Thanks for the answers. I have already fixed it by copying the packages code into the main code and sorting it and now it works 100%.

mmakaay, 2023-02-27:
Wonderful! Enjoy the firmware.

closed 2023-02-27 by mmakaay

stepek, 2023-04-20:
@rafjog can you share some info how you fix this?