mmakaay/esphome-xiaomi_bslamp2 #112: MCU temperature reading support?

szymucha94, 2023-03-16:
Is it safe to use this configuration: https://github.com/esphome/esphome/pull/4330
to send MCU temp to HA from these lamps?

mmakaay, 2023-03-22:
I gave it a go for you, and it seems to work:
[image]

After flashing, my dev lamp did not correctly reboot and I had to disconnect and reconnect the power. I've never seen that happen and likely it has nothing to do with this new module. But in case the same thing happens with you: the power cycle does help ander after restarting, the temperature was reported. Subsequent flashes didn't show this behaviour, so might have been anything really 😄

mmakaay, 2023-03-22:
Great! And interesting you had the same failed reboot. Maybe something that got introduced with 2023.3.0.
I'll experiment a bit with that by doing some downgrade / upgrade cycles. Not important for this issue though.
Enjoy the temperature readings!

closed 2023-03-22 by mmakaay