mmakaay/esphome-xiaomi_bslamp2 #113: Question: Will the (yeelight) App still work?

Madaraxa, 2023-03-20:
Hi.
I found your project because yeelight did cancel the LAN-Control through a Firmware Update.
As I still want to use the yeelight App (mainly to set a Color by clicking on it): is this still possible with your Firmware? Or Do i have to choose between 'Smarthome' and App :)?
I read that the cloud is disabled, but maybe the App still works in local Network?

randybb, 2023-03-20:
I still can control all my Yeelight lamps via LAN, so not sure for which regions they disabled it.

And no, with this integration the lamp will work only with HomeAssistant or with anything that can control devices via MQTT (is not there by default, you can add it).

martinrevert, 2023-03-20:
No, the Yeelight App only works on cloud mode and offer LAN control only for products branded as Yeelight. Maurice's project changes the firmware of bslamps to work on local LAN, so it will not work anymore with Yeelight cloud also. But if you're handy enough, you can add Home Assistant and that will give you more power and control, using de Home Assistant Mobile App companion. I strongly suggest that you explore that scenario if you want keep local control and mobile control of your bslamps.

Madaraxa, 2023-03-20:
I got iobroker, which can handle esphome as well.
I have to use a vis then. Thanks for the answers

closed 2023-03-20 by Madaraxa