mmakaay/esphome-xiaomi_bslamp2 #114: [BUG] Example.yaml won't compile

alexisml, 2023-04-01:
**Describe the bug**
I'm getting the following error:
```
  File "/config/.esphome/external_components/d1d49d2c/components/xiaomi_bslamp2/__init__.py", line 6, in <module>
    from esphome.components.i2c import I2CComponent, I2CDevice
ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/esphome/esphome/components/i2c/__init__.py)
ERROR Unable to import component xiaomi_bslamp2.text_sensor:
```

**To Reproduce**
Steps to reproduce the behavior:
1. Download latest ESPHome 
2. Copy exaple.yaml content
3. Create new device in ESPHome and paste
4. Try to compile

**Expected behavior**
That it compiles :)

**Additional context**
Using latest ESPHome: v2023.3.2 
https://hub.docker.com/r/esphome/esphome

alexisml, 2023-04-01:
I'm now using this one and it works: https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/8f0f2ac290b6257a83ff19409764449cbe2dcf70/example.yaml
However, the one in the repo doesn't: https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/example.yaml

mmakaay, 2023-05-07:
Fun fact: the two example.yaml files that you references are actually the same.
So based on that, it cannot be explained why things started working for you.

mmakaay, 2023-05-11:
As I understand you got things to compile, and it's a bit unclear what happened exactly.
Possibly, you ran into the issues as reported in #116, for which a fix has been released.
I'll close this ticket, assuming that there are no remaining issues for you.

closed 2023-05-11 by mmakaay

Ameb, 2023-05-22:
I think I found the root cause. The example.yaml on the `main` branch. https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/main/example.yaml.

The `dev` branch is the default branch on github, but the [ESPHome site for the device ](https://devices.esphome.io/devices/Xiaomi-Mijia-Bedside-Lamp-2) links to the `main` branch docs.

I got into the repo by first looking at google, then finding the ESPHome site which guided me to the `main` branch docs.