mmakaay/esphome-xiaomi_bslamp2 #116: [BUG] Could not find package with espressif/toolchain-riscv32-esp

TheFes, 2023-05-02:
**Describe the bug**
When I try to update from 2023.4.1 to 2023.4.3 I get the following error as soon as it starts installing the newly compiled firmware:
```
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2
INFO Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2
Error: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2' requirements for your system 'linux_x86_64'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Press the update button, and wait until compiling is completed

**Expected behavior**
Installation of the firmware :)

mmakaay, 2023-05-07:
That is.. well ... Interesting, given that the lamp isn't RISC-V. I will take a look at it.

mmakaay, 2023-05-07:
I tried compiling the firmware using esphome 2023.4.3 and 2023.4.4 from the command line install. I could compile and upload without any issues, so the issue is not a general core issue. Investigating further.

TheFes, 2023-05-08:
Nice, it indeed works like that!

mmakaay, 2023-05-09:
I crafted a new 2023.4.0 release, which should fix the compile issues.

https://github.com/mmakaay/esphome-xiaomi_bslamp2/releases/tag/2023.4.0

mmakaay, 2023-05-11:
Things look good for me with this release. I'll close the issue now.
If there are remaining issues with it, then let me know and I'll reopen the issue.

closed 2023-05-11 by mmakaay