mmakaay/esphome-xiaomi_bslamp2 #117: Retaining state after turning power off and on

cracrama, 2023-05-15:
Hi 
From some time I'm experiencing weird behaviour and I can't pinpoint what is causing it. 
After some ESP/ HA? update device not longer remember it's state after turning power on. (complete power off and on) 

Steps to reproduce the behaviour:
Set brightness/ color, later disconnect device from power - later connect again 
Once device is on, and connected again, it returns to it's default state 100% brightness white mode, instead to it's previous state.  

**Expected behavior**
Retaining state after on off power cycle. 

Please advise 
Kind regards.

mmakaay, 2023-07-11:
Sorry the late answer. I thought I had typed something about this, but apparently not.
I have noticed the same behavior, so I think all lamps will show this behavior at this point.
It is likely that things changed in ESPHome. Possibly this might be fixed by implementing #48 , but I will have to experiment on that.

cracrama, 2023-08-04:
Many thanks for response.
Will wait for new firmware then. 
Best regards.

cracrama, 2024-04-16:
Hi
Sorry im a bit newbie - Can you please explain how to impalement this below solution form issue no 48? 

auto call = the_light.make_call();
call.from_light_color_values(id(global_var));
call.perform();

Please advise 
Best regards