mmakaay/esphome-xiaomi_bslamp2 #120: Issue with flashing - any help please.

jackkitley, 2023-08-20:
**Describe the bug**
Im getting blocked on flashing. Im not sure what to do now.

[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Do this
2. Do that
3. Now this happens

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Trying to flash using a USB - TTL adapter

mmakaay, 2023-08-21:
Normally, issues like this come down to setting up the connections correctly
- RX, TX to the correct pins (try to swap them if not sure)
- GND and GPIO0 must be connected when powering up the lamp, to put it into flash mode

However, in this case I do some boot logs and some chip info, suggesting that there is communication going on between the computer and the ESP32. 
Do you have an option to test the flashing on a windows computer instead? I've seen some people struggle with issues like this when on a mac, and the USB device name looks a bit Mac-y to me.
(side note: I am on a macbook myself, and can flash, so it's not a golden rule that this wouldn't work)

Further than this, I have currently no bright ideas about what could be going on here.

jackkitley, 2023-08-21:
@mmakaay thanks for the reply. i should have updated this. 

It was my USB to TTL. I was using a 2303 and was horrid. TX was 1.8V.

Bought a new one and works great! Lamp working

closed 2023-08-21 by jackkitley

mmakaay, 2023-08-21:
Horrid is bad 😄 
Thanks for the update!