mmakaay/esphome-xiaomi_bslamp2 #121: [BUG] Broken, yellow-ish random light effects

szymucha94, 2023-09-12:
During "slow" and "fast" random light effects sooner or later algorithm picks up the night light color (<2000K white) which breaks the effect - all colors from this point resemble yellow-ish white.

Easy reproduce:
1. Turn on the light with no effects
2. Switch to lowest temperature color (1700K)
3. Enable fast random effect
4. Check output colors

These effects should not be stuck at low temperature.

mmakaay, 2023-09-29:
I have been testing this with my three lamps and haven't seen the effect fail in that way. In fact, I already really often have the lamp in that setup on my desk, since it's fun to have the colors but it's not fun to have them blind me.

How long does is generally take before your lamp gets into this yellow-ish state?
And when you look at home assistant, does the lamp actually switch to white light mode?
I would expect the lamp to be driven to a color, even when it hit white for some reason, since that is what ESPHome's random effect does.

mmakaay, 2023-09-29:
Maybe some logging from the lamp can ... no pun intended ... shine a light on this.

szymucha94, 2023-09-29:
>How long does is generally take before your lamp gets into this yellow-ish state?
It's random. Something between 30 minutes and 3 hours.
>And when you look at home assistant, does the lamp actually switch to white light mode?
Yes. After discovering this I just used an automation as workaround. (when lamp entering white mode and effect is active change mode to rgb)
I'll try to reproduce this issue soon and will provide logs.

mmakaay, 2023-09-29:
One point that I can imagine that makes the lamp use white light mode, is when RGB is (255, 255, 255), so smack in the middle of the color circle. I don't have top of mind if i switch the lamp to white light mode when "white" is selected that way, or if ESPHome might do that. I'll have to dig into the code for that.

My lamp has ben changing colors happily at 2% brightness for > 4 hours now. I'll check back in a few hours to see what it is doing then (it's in my office, and it's dinner time :-) )

mmakaay, 2023-09-29:
Wait, I got my logging turned on, and the color changes are accompanied by:
``` 
[16:10:30][I][light:393]: 'Bedside Lamp Office RGBWW Light' - Keeping current color mode Color temperature for call without color mode.
```

This means that the random colors don't include the color mode as I assumed. So that explains why it would stay in a white light mode when it gets there.
Good news. With this knowledge, I was able to reproduce the behavior easily.

**Reproduce:**

[image]

In the above Home Assistant interface, enable fast random and the lamp will show random colors.
Now click on the white color at the top right of the favorite colors.
After this, the lamp stays in the white light yellow-ish tones.
Clicking a non-white color favorite brings back the random behavior.

**Solution:**

Not sure yet, for that I'll have to do some digging.

mmakaay, 2023-10-02:
In my lamp-specific color mode code, I cannot find anything that would do this RGB -> Temperature color switching.
I did dig into the ESPHome code, and there I find the following code:
https://github.com/esphome/esphome/blob/589b9e10b2c877909083c6551c9d510d02d23ca5/esphome/components/light/light_call.cpp#L385

In this code, a light mode is guessed when no light mode is provided in the light call request. This is the code that logs `Keeping current color mode ...` when the random color changes. There is also a log message that looks like `Using color mode ... for call without color mode`. 

My guess is that it is this ESPHome code that triggers the light mode change. I'm not at home currently, so I can't check the logs right away, but I assume that logging will first show `Keeping current` for RGB messages, then a `Using color` for temperature message and after that only `Keeping current` for template.

If this is the case, the cause is outside of the lamp code and would be an issue to fix in ESPHome code instead.
The most obvious change would likely be to fix the random color effect, which is using old-style calls to change the light's color. When this effect would include a color mode, then this would not happen.

mmakaay, 2025-11-25:
This issue might have been caused by a memory leak error that was fixed a while ago. My development lamp is standard running in slow cycling mode at my desk at home, and I haven't seen the issue anymore since the memory leak fix.

Can you confirm that the issue is indeed fixed using the latest firmware, or does the issue still occur?