mmakaay/esphome-xiaomi_bslamp2 #124: [BUG] Problems with compiling the firmware

lazariv, 2024-02-08:
**Describe the bug**
Compiling fails because custom components cannot be imported.
 
**To Reproduce**
Steps to reproduce the behavior:
1. Use default configuration file
2. esphome compile bedside-lamp.yaml
3. `ImportError: cannot import name 'I2CComponent' from 'esphome.components.i2c' (/home/service/.local/lib/python3.10/site-packages/esphome/components/i2c/__init__.py)`

**Expected behavior**
Expect successful compilation

**Additional context**
Configuration file:
```
# --------------------------------------------------------------------------
# A few practical configuration substitutions.
# --------------------------------------------------------------------------

substitutions:
  name: bedside-lamp
  friendly_name: Bedside Lamp
  transition_length: 500ms

  # Component identifiers.
  prefix: bedside_lamp
  id_light: ${prefix}
  id_light_mode: ${prefix}_light_mode
  id_power_button: ${prefix}_power_button
  id_color_button: ${prefix}_color_button
  id_slider_level: ${prefix}_slider_level
  id_front_panel_illumination: ${prefix}_front_panel_illumination

# --------------------------------------------------------------------------
# Use your own preferences for these components.
# --------------------------------------------------------------------------

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (for captive portal) in case wifi connection fails
  ap:
    ssid: "ESPHome $friendly_name"
    password: !secret wifi_password

captive_portal:

api:
  password: !secret api_encryption_key
  # Disable the reboot timeout. By default, the lamp reboots after 15
  # minutes without any client connections (e.g. when home assistant is off
  # line, or when the WiFi is broken). Reboots are annoying though, because
  # the RGBWW LEDs will turn off during the reboot, causing the light to
  # flicker.
  reboot_timeout: 0s

  # If you want to use light presets (see below) from Home Assistant,
  # then you can expose the required functionality as a service here.
  # This is an example of how you could expose the activation of a preset.
  services:
    - service: activate_preset
      variables:
        my_group: string
        my_preset: string
      then:
        - preset.activate:
            group: !lambda 'return my_group;'
            preset: !lambda 'return my_preset;'

[... 479 lines cut ...]
  for: COLOR_BUTTON
  on_multi_click:
    - timing:
        - ON for at most 0.6s
      then:
        - preset.activate:
            next: preset
    - timing:
        - ON for at least 0.6s
      then:
        - preset.activate:
            next: group
sensor.xiaomi_bslamp2: [source <unicode string>:247]

  Platform not found: 'sensor.xiaomi_bslamp2'.
  platform: xiaomi_bslamp2
  id: bedside_lamp_slider_level
  range_from: 0.02
  on_value:
    then:
      - light.turn_on:
          id: bedside_lamp
          brightness: !lambda |-
            return x;
```

lazariv, 2024-02-08:
The configuration file from `dev`-branch compiles correctly. Is the functionality the same as on `main`?

mmakaay, 2024-02-23:
Sorry for the late answer. Somehow notifications end up in my spam box :-/

The `main` branch can best be ignored. It is an old branch and `dev` is the default branch, which I adopted from the esphome repository.

I dropped the use of `main`, because backward compatibility issues made it a bit hard to keep a clean main branch. To me, `main` communicates "stable!". Therefore, I switched to `dev` to make it clear that this is the development branch, that might sometimes be broken (I try not to break it though, all my home lamps are running this ;-) )
For releases, I branch `dev` to a specific `release/<version>` branch and publish that one as the release version.

As for what repository to use: use `dev` if you are feeling lucky and trust in my competence, otherwise stick with the latest release version, which is currently [2023.4.0](https://github.com/mmakaay/esphome-xiaomi_bslamp2/tree/2023.4.0). 

This might sound like a really old version, but fact is that it works well and it is still compatible with the current ESPHome version. I am working on a new version, with mainly changes in the build flags that make for a cleaner more standard build (the Espressif framework now has new features that can be used to clean things up). This means there will be a 2024 release in the upcoming future. But still, nothing too big feature-wise.

I hope this clears up your question.
Enjoy your lamp!

mmakaay, 2024-02-24:
BTW: keeping this issue open, as a reminder to remove the main branch to prevent others from running into it.

mmakaay, 2024-10-22:
The main branch has been removed after releasing 2024.10.0, so closing the issue.

closed 2024-10-22 by mmakaay