mmakaay/esphome-xiaomi_bslamp2 #125: Reconfigure lamp

iAbadia, 2024-02-29:
Context:
Hi, I followed this project a couple years ago (thank you so much for your work!) and since then I've changed my network configuration a couple times, always keeping the same BSSID and password. Every time the lamp has re-connected just fine but I'm now unable to reproduce the same conditions and I need to re-configure the lamp.

The actual question:
Is there a way to reconfigure the lamp without opening it up again?

helgek, 2024-03-10:
You can add the lamp to ESPHome dashboard by creating a generic ESP32 device. Then you edit the yaml of this device and you overwrite it with the yaml config of your bedside lamp. Now the lamp should show up in the ESPHome dashboard. You can now adjust the wifi settings and install the configuration which will overwrite the current firmware of your lamp.

mmakaay, 2024-10-18:
It depends a bit on where you are at now. When the BSSID was changed, and the lamp can therefore no longer connect to the WiFi network, it will not be able to present itself to the ESPHome dashboard.

If you can still connect the lamp to the old BSSID, you can still flash it. In that case, you could flash it with the new BSSID and then let it connect to the new network. Another suggestion would be to configure the captive portal feature in the device YAML (see: [https://esphome.io/components/captive_portal](https://esphome.io/components/captive_portal)). With this feature, the device exposes a WiFi access point when connecting to WiFi fails. That would give you a way to reconfigure the device when the network changes.

Based on your question, it sounds like you switched to a new BSSID, and the old BSSID is no longer available. In that case, flashing can only be done by connecting the lamp to a serial port. Unless of course, you can temporarily setup a WiFi network with the old connection settings, giving you the option to do an OTA flash operation.