mmakaay/esphome-xiaomi_bslamp2 #126: [BUG] Not possible to set the bslamp2 into warm light

helgek, 2024-03-10:
**Describe the bug**
In comparision with the original firmware I'm not able to get a light color reproduction of 1700K in CCT mode.

**To Reproduce**
Activate CCT mode and chose Kelvin color setting in the range of 1700K-2000K.

**Expected behavior**
Warm light recognisably representing the 1700K-2000K range.

**Additional context**
Please see the following comparison of ESPHome firmware and original firmware in pictures:

[image]

[image]

[image]

[image]

[image]

mmakaay, 2024-10-18:
What I've seen with other devices too, is that there seems to be some calibration required on the hardware in order to get the same results as the original firmware. The ways the outputs are driven now are derived from me measuring at many (MANY!) different light settings, and building lookup tables out of those to find and interpolate settings. Unfortunately, the output levels looked nothing like clean linear RGBWW output values.

Since there are more reports of light outputs not matching the original outputs, I think that some additional parameters for calibrating the light are required. For example a scaling factor for every channel, that linearly changes the output level for those channels. That will require some experimentation to see if that is a feasible work-around here. I will look into it to see if I can come up with something.

For the night light output levels, I know that some people tweaked the output levels in code to get their desired output. Even to tweak the night light color to a desired color, e.g. blue. That would probably not work for your case, since you also have issues with the non-nightlight settings. Updating the whole output value table is not something I would recommend. It was painful enough, going from measurements to a table. Since the relation between output levels and resulting light color are a bit weird, changing values to get the desired light output is likely quite the ordeal.

vojtechj, 2025-01-09:
and could this issue be related to the problem that if I set yellow color in home assistant the lamp outputs something between yellow and green .... other colors look good.

helgek, 2025-01-09:
[fwbackup_bedside_lamp_2.bin.zip](https://github.com/user-attachments/files/18368441/fwbackup_bedside_lamp_2.bin.zip)

just adding the original firmware in case it could be useful

mmakaay, 2025-11-24:
Last week, I have tried calibration ideas, but the enormously non-linear nature of the LED output levels is working against us quite a bit here :-/ 

I even threw some AI at it, to see if it could find secret computations for mapping RGBWW values to the output values to use. I got many happy reports about finding working computations that resulted in output values that differed max 2% with my output level tables. That was all massive hallucination though. Sure, the first three values were correct, but I got to 2500%+ differences when going further up the RGBWW value scales :-D

I can take a look at the original firmware to see if there's anything to reverse engineer from there.

rkoptev, 2026-08-16:
Any updates on the issue?

mmakaay, 2026-08-17:
The last update still stands. We have a calibration feature for the night light. For the warm white light difference, I have no ideas how to change the behavior in a generic way that works on all devices.

rkoptev, 2026-08-17:
Maybe I'm doing something wrong, but no matter how I adjust the 3 settings for the night light (level, temperature, tint) - light color and temperature stays the same. I tried using a calibration table from @ArtikusHG - but it yields reddish and still too bright night light. What I have on default calibration is exactly what @helgek showed on his pictures. The difference is night and day (yep, a silly pun). It's not a bit off, it is order of magnitude brighter, and more like 4000K on the minimal setting compared to the original.