mmakaay/esphome-xiaomi_bslamp2 #127: [BUG] Can't get into Nightlight Mode with physical slider

helgek, 2024-03-10:
**Describe the bug**
When I go to the lowest setting with the physical slider I'm not able to activate the 1% / Nightlight setting.

khmm12, 2024-05-04:
It’s not a bug because that’s how supposed to be. You can exclude the default behavior configuration [package](https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/2a6944e2f9990fbea9a58d87bb57877ec607512c/example.yaml#L38), then copy-paste changing the minimum slider value [here](https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/2a6944e2f9990fbea9a58d87bb57877ec607512c/packages/behavior_default.yaml#L156).

helgek, 2024-06-28:
@khmm12 Thank you! I will look into this later and try my luck :)

closed 2025-03-03 by mmakaay