mmakaay/esphome-xiaomi_bslamp2 #130: [BUG] esphome 2024.10.0 Base MAC address from BLK3 of EFUSE version error

b2un0, 2024-10-16:
**Describe the bug**
after update esphome to 2024.10 i got the following errors and the lamp dies not comes up or is useable.
also all the lamp comes up as a "new" device and no entities are shown.

```
esp-idf:000]: E (510963) system_api: Base MAC address from BLK3 of EFUSE version error, version = 74
```

**To Reproduce**
Steps to reproduce the behavior:
1. update esphome to 2024.10.0
2. read the logs

**Expected behavior**
a working BSLamp2 with esphome

randybb, 2024-10-16:
There were multiple changes related to EFUSE in esphome mentioned in this thread https://discord.com/channels/429907082951524364/1290562513565384808

Basically I am keeping all possible ignores in my config to stay on the safe side:
```yaml
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      ESP_MAC_IGNORE_MAC_CRC_ERROR: y
    advanced:
      ignore_efuse_mac_crc: true
      ignore_efuse_custom_mac: true
```

b2un0, 2024-10-16:
> There were multiple changes related to EFUSE in esphome mentioned in this thread https://discord.com/channels/429907082951524364/1290562513565384808

this link is useless without knowing the related discord server :D

b2un0, 2024-10-16:
esphome/esphome-docs/pull/4297

TomaszDom, 2024-10-17:
what do we do now? have to solder leads again to reflash all the lamps? but there's no immediate fix in the dev branch yet, correct?

BlairC1, 2024-10-17:
mine came back online in esphome after a power cycle, working fine after adding the config from randybb.

ezcGman, 2024-10-17:
> what do we do now? have to solder leads again to reflash all the lamps? but there's no immediate fix in the dev branch yet, correct?

you can use my branch as reference, or manually just add just this (no need for the full block, as randybb posted above):
```
esp32:
  framework:
    advanced:
      ignore_efuse_mac_crc: true
      ignore_efuse_custom_mac: true
```

randybb, 2024-10-18:
yes, ignore_efuse_mac_crc is setting ESP_MAC_IGNORE_MAC_CRC_ERROR, so this is not needed. the other ignore is only thing which you need to add to your config. After that esphome will accept MAC address stored in EFUSE.
If you have more lamps just disconnect all lamps as these are sharing now the same MAC, then power one lamp, update it from the right config, then you can power another one,...

TomaszDom, 2024-10-18:
thanks, it worked. indeed I have >1 lamp.

mmakaay, 2024-10-18:
Thanks for stepping in and helping so far @randybb ! 
I am away from home at the moment. Sunday, I'll get back and I will update the bslamp project to make the built-in config package compatible with the recent ESPHome changes.

mmakaay, 2024-10-22:
I released version 2024.10.0 of the firmware, which addresses the reported issue.
See https://github.com/mmakaay/esphome-xiaomi_bslamp2/releases/tag/2024.10.0 for the releaes notes.

mmakaay, 2024-10-23:
I got some reports on the new firmware working correctly with ESPHome 2024.10.0, so closing this ticket now.

closed 2024-10-23 by mmakaay